WPIRoboticsProjects/GRIP #525: Grip GUI on the Raspberry Pi 2

tomatac, 2016-02-24:
Can Grip GUI run on the Raspberry Pi 2, so no deployment would be necessary?
We have issues deploying on the RoboRio and I understand the there are similar issues deploying to the Raspberry Pi. After deployment the camera does not send good image for vision processing. See issue #494,

JLLeitschuh, 2016-02-25:
Duplicate of issue #389
Oracle will not be supporting JavaFX in the arm embedded JRE
https://github.com/WPIRoboticsProjects/GRIP/issues/389#issuecomment-172973654

tomatac, 2016-02-25:
Did anybody tried OpenJFX?
https://wiki.openjdk.java.net/display/OpenJFX/Main

JLLeitschuh, 2016-02-25:
We welcome anyone trying to get this to work. Please feel free to open a pull request if you do.
Come chat with us in gitter if you have any ideas. https://gitter.im/WPIRoboticsProjects/GRIP

rich-coe, 2016-08-15:
Here are my notes for building openjfx on arm:
from: https://wiki.openjdk.java.net/display/OpenJFX/Building+OpenJFX

install deps
````
    sudo apt-get install bison flex gperf libasound2-dev libgl1-mesa-dev \
        libgstreamer0.10-dev libgstreamer-plugins-base0.10-dev libjpeg-dev \
        libpng-dev libx11-dev libxml2-dev libxslt1-dev libxt-dev \
        libxxf86vm-dev pkg-config x11proto-core-dev \
        x11proto-xf86vidmode-dev  libgtk2.0-dev \
        libxtst-dev libudev-dev 

    edit /etc/apt/sources.list
        and uncomment universe sources

    sudo apt-get update

    sudo apt-get install mercurial libavcodec-dev libavformat-dev
````

install gradle
        `wget https://services.gradle.org/distributions/gradle-1.8-bin.zip`

Apply Diffs 
````
diff --git a/build.gradle b/build.gradle
index c1d2486..843d99a 100644
--- a/build.gradle
+++ b/build.gradle
@@ -704,7 +704,7 @@ if (IS_WINDOWS && OS_ARCH != "x86" && OS_ARCH != "amd64") {
     throw new Exception("Unknown and unsupported build architecture: $OS_ARCH")
 } else if (IS_MAC && OS_ARCH != "x86_64") {
     throw new Exception("Unknown and unsupported build architecture: $OS_ARCH")
-} else if (IS_LINUX && OS_ARCH != "i386" && OS_ARCH != "amd64") {
+} else if (IS_LINUX && OS_ARCH != "i386" && OS_ARCH != "amd64" && OS_ARCH != "arm") {
     throw new Exception("Unknown and unsupported build architecture: $OS_ARCH")
 }

@@ -2554,8 +2554,8 @@ project(":web") {
                     } else {
                         environment([
                             "QMAKESPEC"      : "linux-g++-32",
-                            "QMAKE_CFLAGS"   : "-m32",
-                            "QMAKE_LFLAGS"   : "-m32",
+                            // "QMAKE_CFLAGS"   : "-m32",
+                            // "QMAKE_LFLAGS"   : "-m32",
                         ])
                     }
                 } else if (t.name.startsWith("arm")) {
@@ -3260,8 +3260,8 @@ project(":apps") {
               }
               ant.properties['jfxbuild.jfxrt.jar'] = jfxrtJar
               ant.properties['platforms.JDK_1.8.home'] = "${rootProject.ext.JDK_HOME}"
-              ant.project.executeTarget("sampleAppsJar")
-              ant.project.executeTarget("scenebuilderSampleAppsJar")
+              // ant.project.executeTarget("sampleAppsJar")
+              // ant.project.executeTarget("scenebuilderSampleAppsJar")
               if (!t.name.startsWith("arm")) {
                 ant.project.executeTarget("scenebuilderAppJar")
               }
````
````
diff --git a/buildSrc/linux.gradle b/buildSrc/linux.gradle
index 753d79c..34a313f 100644
--- a/buildSrc/linux.gradle
+++ b/buildSrc/linux.gradle
@@ -45,7 +45,7 @@ def commonFlags = [
         "-fno-strict-aliasing"
    "-fPIC", "-fno-omit-frame-pointer", // optimization flags
         "-W"
    "-Wall", "-Wno-unused", "-Wno-parentheses", "-Werror=implicit-function-declaration"] // warning f

-if (!IS_64) {
+if (false && !IS_64) {
     commonFlags += "-m32"
 }

@@ -173,7 +173,7 @@ LINUX.launcher.compiler = compiler
 LINUX.launcher.ccFlags = ["-DJAVAARCH=\"$OS_ARCH\""
    "-I$JDK_HOME/include", "-I$JDK_HOME/include/linux", "-c"]
 LINUX.launcher.linker = linker
 LINUX.launcher.linkFlags = ["-ldl"]
-if (!IS_64) {
+if (false && !IS_64) {
     LINUX.launcher.ccFlags += "-m32"
     LINUX.launcher.linkFlags += "-m32"
 }
@@ -183,7 +183,7 @@ LINUX.launcherlibrary.compiler = compiler
 LINUX.launcherlibrary.ccFlags = ["-DJAVAARCH=\"$OS_ARCH\""
    "-I$JDK_HOME/include", "-I$JDK_HOME/include/linux",
 LINUX.launcherlibrary.linker = linker
 LINUX.launcherlibrary.linkFlags = ["-ldl"
    "-lpthread", "-shared"]
-if (!IS_64) {
+if (false && !IS_64) {
     LINUX.launcherlibrary.ccFlags += "-m32"
     LINUX.launcherlibrary.linkFlags += "-m32"
 }
````

Build javafx
        `gradle`

Test javafx
        `gradle test`

Package
        `gradle zips`

Extract jar
        `build/bundles/javafx-sdk-overlay.zip contains data to overlay existing JAVA_HOME`

JLLeitschuh, 2016-08-21:
Can we run the UI tests against the open JDK?

You may run into an issue because we import the Sun implementation libraries of JavaFX in places to prevent dreadlocks. Places where we use `PlatformImpl.runAndWait` could be problematic.