WPIRoboticsProjects/GRIP #526: Running GRIP within same JVM as FRC User Program

JLLeitschuh, 2016-02-24:
So I tried this bit of code:
```java
@Override
protected void robotInit() {
	new Thread(() -> {
		final File f = new File("/home/lvuser/grip.jar");
		try {
			URLClassLoader child = new URLClassLoader (new URL[] { f.toURI().toURL()}, this.getClass().getClassLoader());
			Class classToLoad = Class.forName ("edu.wpi.grip.core.Main", true, child);
        	Method method = classToLoad.getDeclaredMethod ("main", String[].class);
        	Object result = method.invoke (null, new Object[] {new String[]{"/home/lvuser/project.grip"}});
		} catch (MalformedURLException | ClassNotFoundException | NoSuchMethodException | InvocationTargetException | IllegalAccessException e) {
			throw new AssertionError("Failed to launch GRIP", e);
		}	
	}, "GRIP Thread").start();
	
}
```

It does launch successfully, but there are two problems.

1. We effectively take over the default thread handler for the entire JVM which means that if any threads die our handler will run instead of whatever someone has defined for the FRC User Program.
2. Network Tables tries to connect as a client instead of as a server meaning that it can't connect as it's in the same process.

Also, the FRC user program is now even larger.
```
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
15790 15787 lvuser   S     225m  92%  78% /usr/local/frc/JRE/bin/java -jar /home/lvuser/FRCUserProgram.jar
```

ThomasJClark, 2016-02-24:
> We effectively take over the default thread handler for the entire JVM which means that if any threads die our handler will run instead of whatever someone has defined for the FRC User Program.

AFAIK wpilib doesn't set a default exception handler.  It should be fine if we override it, as long as exceptions ultimately get logged.

> Network Tables tries to connect as a client instead of as a server meaning that it can't connect as it's in the same process.

Maybe if networktables is already connected when we start up, we can just not try to do any additional configuration.

JLLeitschuh, 2016-02-24:
> AFAIK wpilib doesn't set a default exception handler. It should be fine if we override it, as long as exceptions ultimately get logged.

Our handler intentionally stops the entire JVM in some circumstances.

ZeusChops, 2016-03-23:
The 'best' fix is removing "-XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError" on deploy that way the crash is only handled within GRIP instead of the entire JVM.

JLLeitschuh, 2016-03-24:
That won't fix the problem.

ThomasJClark, 2016-03-24:
I guess you could just do `Thread.setDefaultUncaughtExceptionHandler(() -> {});` after loading GRIP.

JLLeitschuh, 2016-03-24:
Yea, that would work.
Or just pass it null.

AustinShalit, 2016-07-29:
We should put this in the Wiki article.

SamCarlberg, 2016-10-11:
This isn't needed any more since code generation has been merged.

closed 2016-10-11 by SamCarlberg