WPIRoboticsProjects/GRIP #528: GRIP Fails to start on ubuntu

TechplexEngineer, 2016-02-28:
Just installed the latest release of GRIP (FRC Update 3)

Tried to run GRIP via the command line and got this error:
```
/opt/GRIP $ ./GRIP 
Feb 28, 2016 12:23:22 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Feb 28, 2016 12:23:22 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.2.0-1-g0b93b8a
platform: /Linux/amd64/
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM
```

It looked like @multiplemonomials had success using the $ java -jar ui-1.2.0.jar in issue #345 

```
/opt/GRIP/app $ java -jar ui-1.2.0.jar 
Feb 28, 2016 12:24:53 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Feb 28, 2016 12:24:53 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.2.0-1-g0b93b8a
platform: /Linux/amd64/
Exception in Application start method
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:767)
Caused by: java.lang.RuntimeException: Exception in Application start method
	at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:917)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$155(LauncherImpl.java:182)
	at java.lang.Thread.run(Thread.java:745)
Caused by: javafx.fxml.LoadException: 
file:/opt/GRIP/app/ui-1.2.0.jar!/edu/wpi/grip/ui/MainWindow.fxml:17

	at javafx.fxml.FXMLLoader.constructLoadException(FXMLLoader.java:2601)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2579)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:3214)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:3175)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:3165)
	at edu.wpi.grip.ui.Main.start(Main.java:61)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$162(LauncherImpl.java:863)
	at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$175(PlatformImpl.java:326)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$49(GtkApplication.java:139)
	... 1 more
Caused by: com.google.inject.ProvisionException: Unable to provision, see the following errors:

1) Error injecting constructor, java.lang.UnsatisfiedLinkError: /tmp/libNetworkTablesJNI419331451862598523.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /tmp/libNetworkTablesJNI419331451862598523.so)
  at edu.wpi.grip.core.operations.networktables.NTManager.<init>(Unknown Source)
  at edu.wpi.grip.core.operations.networktables.NTManager.class(Unknown Source)
  while locating edu.wpi.grip.core.operations.networktables.NTManager
    for field at edu.wpi.grip.core.Pipeline.ntManager(Unknown Source)
  at edu.wpi.grip.core.Pipeline.class(Unknown Source)
  while locating edu.wpi.grip.core.Pipeline
    for field at edu.wpi.grip.ui.MainWindowController.pipeline(Unknown Source)
[... 699 lines cut ...]
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$49(GtkApplication.java:139)
	at java.lang.Thread.run(Thread.java:745)

4 errors
	at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1025)
	at com.google.inject.internal.InjectorImpl.getInstance(InjectorImpl.java:1051)
	at javafx.fxml.FXMLLoader$ValueElement.processAttribute(FXMLLoader.java:929)
	at javafx.fxml.FXMLLoader$InstanceDeclarationElement.processAttribute(FXMLLoader.java:971)
	at javafx.fxml.FXMLLoader$Element.processStartElement(FXMLLoader.java:220)
	at javafx.fxml.FXMLLoader$ValueElement.processStartElement(FXMLLoader.java:744)
	at javafx.fxml.FXMLLoader.processStartElement(FXMLLoader.java:2707)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2527)
	... 14 more
Exception running application edu.wpi.grip.ui.Main
```

```
$ java -version
java version "1.8.0_72"
Java(TM) SE Runtime Environment (build 1.8.0_72-b15)
Java HotSpot(TM) 64-Bit Server VM (build 25.72-b15, mixed mode)
```

multiplemonomials, 2016-02-28:
Bottom line: Linux ABIs are a fantastically broken mess, even across versions of the same distro.  

Maybe it's time to reimplement NetworkTables in pure Java, or just have users build it themselves with CMake.

multiplemonomials, 2016-02-28:
The temporary fix would be to recompile networktables with a much older compiler and standard library, which is available on more devices.  (GCC 4.7 perhaps)

JLLeitschuh, 2016-02-29:
This problem is detailed here.
You have to update GCC.
https://github.com/WPIRoboticsProjects/GRIP/wiki/Setting-up-build-tools#ubuntu-network-tables-throws-a-bunch-of-exceptions

AustinShalit, 2016-04-23:
@TechplexEngineer Has this issue been solved? If so, please close this issue. If not, please let us know how we can help.

AustinShalit, 2016-07-29:
I am going to close this issue because it has grown stale.  Please reopen the issue if you are still having trouble.

closed 2016-07-29 by AustinShalit