WPIRoboticsProjects/GRIP #535: If camera isn't connected, GRIP GUI pop-up spams the environment with UnknownHostException

ryannazaretian, 2016-03-06:
If the wrong address is input or the camera isn't connected, GRIP will throw the UnknownHostException over and over again. Clicking cancel will close the exception, but GRIP will pop the exception back up immediately. 

> Thread[IP Camera raspberrpi.local Service,5,main] threw an exception
## Stack Trace:
```java
java.lang.RuntimeException: edu.wpi.grip.core.sources.GrabberService$GrabberServiceException: [IP Camera raspberrpi.local] Failed to start
	at com.google.common.base.Throwables.propagate(Throwables.java:160)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:77)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:95)
	at java.lang.Thread.run(Thread.java:745)
Caused by: edu.wpi.grip.core.sources.GrabberService$GrabberServiceException: [IP Camera raspberrpi.local] Failed to start
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:51)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:55)
	... 2 more
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: raspberrpi.local
	at edu.wpi.grip.core.sources.IPCameraFrameGrabber.start(IPCameraFrameGrabber.java:99)
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:49)
	... 3 more
Caused by: java.net.UnknownHostException: raspberrpi.local
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1169)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1105)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:999)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:933)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1513)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1441)
	at edu.wpi.grip.core.sources.IPCameraFrameGrabber.start(IPCameraFrameGrabber.java:96)
	... 4 more

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.3.0-rc1-9-g48dd139
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 10
os.version | 10.0
os.arch | amd64

ryannazaretian, 2016-03-06:
Also, there is a substantial memory leak when the camera isn't found.

JLLeitschuh, 2016-03-14:
Oh.... Well then, thats not what I expected to happen.
Thanks for opening this.

JLLeitschuh, 2016-03-14:
Uhhhhhh.... Well... That exception shouldn't be there.
We are running guava 19 which doesn't have that code anymore.
I have no idea why, but apparently windows is using an older version of guava.

JLLeitschuh, 2016-03-14:
Are you building/running from source?

ryannazaretian, 2016-03-15:
Installed through the release installer.

AustinShalit, 2016-08-10:
The newest release (1.5.0 solves this build issue)

closed 2016-08-10 by AustinShalit