WPIRoboticsProjects/GRIP #536: Using Grip while on FMS no information publishing to network table

chris2852, 2016-03-06:
When we are tethered or on our own wireless network we have no issues all the number on the network table show up.
When we are attached to the field the network table shows only nulls '[]' in all the options.
Sometimes when we open the network table it only shows Root and nothing else.
Every time we have tested anything we made sure Grip was processing the image correctly and only had 1 contour found on the last image. 
We have always been able to have image feedback to our dashboard.

Anyone else have this issue? Any information would really help

bradamiller, 2016-03-07:
Are you running GRIP on a co-processor like the raspberry pi? If so it may be a networking issue. When you are tethered or on your network, mDNS is used to resolve IP addresses to the robot (network tables server). On the field DNS is used, that's new this year. There may be an issue with the GRIP client finding the roboRIO network tables server on the roboRIO. You can use static IP addresses everywhere using the 2014 numbering scheme (10.te.am.2 for robot, 10.te.am.5 for laptop, and 10.te.am.11 for a network camera). If you are using a co-processor, then use 10.te.am.19 or some high number like that to stay out of the preallocated range that the field gives out.

chris2852, 2016-03-08:
Hi Brad,

We are running GRIP on our Driver station.
I am not a big network guy i understand the basics but i will pass this information on to my programming team and see if this makes sense for them.
But if i remember from the past weekend we have static on the camera but not sure about the laptop or robot.

robert-peters, 2016-03-08:
Brad,

We are running GRIP on our driver station.  In the settings we originally had "roborio-2852-frc.local" which worked at our practice field or when tethered at the event but when we connected to FMS the Network Table did not get updated even though GRIP found the contours we were looking for.  We tried using "roborio-2852-frc.local:5800" but that did not work either.  Is there a syntax that I could use that references the roboRIO using its static IP?

bradamiller, 2016-03-08:
I’m not sure and hope that one of the GRIP guys will respond to this. Last year mDNS was used whether you were at home, or tethered, or on the field. This year, to speed up the name lookups on the field, there is a DNS server running as part of the FMS and so the names no longer have the .local after them. One solution is to just use static IPs on the field and at home, but it has the downside of having to change the IP address on the laptop from static to DHCP when you are trying to get internet access.

To use static IPs use the pre-2015 numbering scheme - make your robot 10.28.52.2, the driver station 10.28.52.5, and a network camera (if yo have one) 10.28.52.11. If it turns out that GRIP doesn’t natively support static IP addresses, you can cheat by changing the hosts file in windows and adding an entry for roborio-2852-frc.local with the address 10.28.52.2. This web page:
http://www.thewindowsclub.com/hosts-file-in-windows
shows how to edit the file. The examples they give are about blocking sites, but really the hosts file is just a mapping from computer names to IP addresses that can override the normal lookups on windows. So putting a line in there for your roboRIO will cause windows to use your provided IP address when it sees the robot name.

Brad

On March 8, 2016 at 8:12:14 AM, robert-peters (notifications@github.com) wrote:

We are running GRIP on our driver station. In the settings we originally had "roborio-2852-frc.local" which worked at our practice field or when tethered at the event but when we connected to FMS the Network Table did not get updated even though GRIP found the contours we were looking for. We tried using "roborio-2852-frc.local:5800" but that did not work either. Is there a syntax that I could use that references the roboRIO using its static IP?

robert-peters, 2016-03-08:
Brad,

I apologize, I am not a networking guy.  If I adjust the hosts file, do I use the following syntax to push the data through the FMS allowed pipelines?

10.28.52.2:5800   roborio-2852-frc.local

Rob

Robert Peters, P.Eng.
President
Mobile: 416-580-3145
[CallistoMech for Light BG]

http://www.thewindowsclub.com/hosts-file-in-windows

This communication is for use by the intended recipient and contains information that may be Privileged, confidential or copyrighted under applicable law. If you are not the intended recipient, you are hereby formally notified that any use, copying or distribution of this e-mail, in whole or in part, is strictly prohibited. Please notify the sender by return e-mail and delete this e-mail from your system.

bradamiller, 2016-03-08:
That’s correct except you should leave off the port number (5800). If I’m understanding your issue correctly you are trying to communicate to the robot using network tables and the FMS network configuration already allows that. So you if just use:

10.28.52.2   roborio-2852-frc.local

Then any lookups on the network address for roborio-2852-frc.local will automatically resolve to 10.28.52.2 which is the roborio address.

Brad

http://www.thewindowsclub.com/hosts-file-in-windows

robert-peters, 2016-03-08:
Okay, I will give it a try.

Robert Peters, P.Eng.
President
Mobile: 416-580-3145
[CallistoMech for Light BG]

http://www.thewindowsclub.com/hosts-file-in-windows

robert-peters, 2016-03-08:
Brad,

I will be at the GTA East event in Oshawa, Ontario on Thursday morning to test the communications.  Is there a way that I can get in touch if the suggestions that you have made do not work?

Rob

Robert Peters, P.Eng.
President
Mobile: 416-580-3145
[CallistoMech for Light BG]

http://www.thewindowsclub.com/hosts-file-in-windows

bradamiller, 2016-03-08:
Sure. You can send email. I’ll be at the WPI District Competition over the weekend starting on Friday. 

Brad

http://www.thewindowsclub.com/hosts-file-in-windows

robert-peters, 2016-03-10:
Brad,

We resolved the issue.  I could not get the host file redirection to work but I found out that although I can connect to the Network Table by using roborio-2852-frc.local when not connected to the field, I needed to use roboRIO-2852-FRC.local when connected.  Once I entered the address into GRIP using the correct case, GRIP was able to update the Network Table and the Rio could see the results.

Perhaps a new release of GRIP is required since it defaults to the lower case version of the address which is probably giving teams issues even if they  are trying to deploy GRIP onto their Rio.

Thanks for your help.

Robert Peters, P.Eng.
President
Mobile: 416-580-3145
[CallistoMech for Light BG]

http://www.thewindowsclub.com/hosts-file-in-windows

AustinShalit, 2016-04-23:
@chris2852 @robert-peters Has this issue been solved? If so, please close this issue. If not, please let us know how we can help.

PR #645 (2016-08-01): Update default roboRIO mDNS address
closes #536 

Changes the default roboRIO mDNS address to match capitalization as discussed in #536.
closed 2016-08-01 by JLLeitschuh