WPIRoboticsProjects/GRIP #538: 0 contours found

balloman, 2016-03-13:
I am going through the tutorial for the 2016 game, but when I get to the step where I have to filter contours, I get 0 results each time, no matter what the values I put in there. I even put in 1 as the minimum area and still got nothing

bradamiller, 2016-03-13:
The sample programs in screensteps and the wiki are based on the set of images taken before kickoff with our camera and ring light. The color of your ring light, camera settings, and lens focal length will change the parameters you need to detect goals.

balloman, 2016-03-13:
I am using those images, but I ended up reverting to the latest release instead of using the pre-release, and that fixed the problem

JLLeitschuh, 2016-03-14:
Can you send us an example picture and your save. Perhaps this is a bug that we can fix.

randomtestfive, 2016-03-15:
I'm having this problem too. This is using the default filter contour settings
[image]

ThomasJClark, 2016-03-15:
What version of GRIP? 

Can you post a screenshot of the entire filter step? (just to make sure the defaults are really what we think)

randomtestfive, 2016-03-15:
[image]

balloman, 2016-03-15:
This seems to only happen on the pre-release as the latest release has no problem

Sam-Carlberg, 2016-03-18:
It's a bug with Max Vertices. Seems like that's defaulting to zero (or some other small number) instead of the number in the box. Just changing that number will make it work.

JLLeitschuh, 2016-03-19:
I wonder what's causing the bug in `Max Vertices` is.

Sam-Carlberg, 2016-03-19:
Yeah, it's defaulting to 0. Should either just set it to 1,000,000 or put checks in the builder to set the initial value to `low` or `high` if it's out of bounds.

closed 2016-03-20 by JLLeitschuh

ryannazaretian, 2016-03-28:
Can also confirm that I had this issue this weekend when testing GRIP. Version 1.3 RC2

JLLeitschuh, 2016-04-11:
This should be fixed in the newest release.

PR #548 (2016-03-20): Fixed Max Vertices defaulting to zero and displaying 1,000,000
Fixes #538. The default value and minimum value were switched.