WPIRoboticsProjects/GRIP #543: Dead lock issue while trying to quit the application

calper95, 2016-03-14:
Hi, today I encountered a problem with the app while trying to exit. Here is what I did:

I opened the camera input and directed it to the CV resize. Then I directed the output of resize to Canny algorithm. After that I directed the output of the Canny algorithm to the Find Contours. Then I directed the camera output directly to the Find Lines function.

I had the outputs displaying on the Preview whenever I linked an input to the functions.

After these I tried to quit the application by selecting quit in the file tab. This is when the application asked me for "Save Project?". I selected the no option and the application went into a lock and become unresponsive. No error messages was displayed whatsoever.

**# Thread Dump:**

```
2016-03-14 17:30:20
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.66-b17 mixed mode):

"RMI TCP Connection(2)-130.215.10.213" #33 daemon prio=9 os_prio=31 tid=0x00007f9f96d01000 nid=0x15877 runnable [0x0000700002822000]
   java.lang.Thread.State: RUNNABLE
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:170)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:265)
	- locked <0x0000000795c5c650> (a java.io.BufferedInputStream)
	at java.io.FilterInputStream.read(FilterInputStream.java:83)
	at sun.rmi.transport.tcp.TCPTransport.handleMessages(TCPTransport.java:550)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run0(TCPTransport.java:826)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.lambda$run$256(TCPTransport.java:683)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler$$Lambda$649/1283470175.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run(TCPTransport.java:682)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- <0x0000000795660b20> (a java.util.concurrent.ThreadPoolExecutor$Worker)

"JMX server connection timeout 32" #32 daemon prio=9 os_prio=31 tid=0x00007f9f96b12000 nid=0x16753 in Object.wait() [0x0000700002720000]
   java.lang.Thread.State: TIMED_WAITING (on object monitor)
	at java.lang.Object.wait(Native Method)
	at com.sun.jmx.remote.internal.ServerCommunicatorAdmin$Timeout.run(ServerCommunicatorAdmin.java:168)
	- locked <0x00000007957a5580> (a [I)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- None

"RMI Scheduler(0)" #31 daemon prio=9 os_prio=31 tid=0x00007f9f96ff2000 nid=0x17d7b waiting on condition [0x000070000261d000]
   java.lang.Thread.State: TIMED_WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000795638b78> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2078)
	at java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:1093)
	at java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:809)
	at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1067)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1127)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- None

"RMI TCP Connection(1)-130.215.10.213" #30 daemon prio=9 os_prio=31 tid=0x00007f9f926ac800 nid=0xe92b runnable [0x0000700002519000]
   java.lang.Thread.State: RUNNABLE
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:170)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
[... 406 lines cut ...]
   Locked ownable synchronizers:
	- None

"VM Thread" os_prio=31 tid=0x00007f9f91005000 nid=0x4203 runnable 

"GC task thread#0 (ParallelGC)" os_prio=31 tid=0x00007f9f9200b000 nid=0x3203 runnable 

"GC task thread#1 (ParallelGC)" os_prio=31 tid=0x00007f9f91800800 nid=0x3403 runnable 

"GC task thread#2 (ParallelGC)" os_prio=31 tid=0x00007f9f91803000 nid=0x3603 runnable 

"GC task thread#3 (ParallelGC)" os_prio=31 tid=0x00007f9f91804000 nid=0x3803 runnable 

"GC task thread#4 (ParallelGC)" os_prio=31 tid=0x00007f9f91804800 nid=0x3a03 runnable 

"GC task thread#5 (ParallelGC)" os_prio=31 tid=0x00007f9f91805000 nid=0x3c03 runnable 

"GC task thread#6 (ParallelGC)" os_prio=31 tid=0x00007f9f91805800 nid=0x3e03 runnable 

"GC task thread#7 (ParallelGC)" os_prio=31 tid=0x00007f9f91806800 nid=0x4003 runnable 

"VM Periodic Task Thread" os_prio=31 tid=0x00007f9f90837000 nid=0x5603 waiting on condition 

JNI global references: 4630
```

ZeusChops, 2016-03-23:
What operating system is this on? For clarification, this was during GUI runtime and not deployment runtime correct?

JLLeitschuh, 2016-03-24:
It was a Mac.
This was with the GUI.
They were doing File -> Exit

AustinShalit, 2016-08-10:
Related: #660

SamCarlberg, 2017-01-29:
This should have been resolved with #689. Re-open this PR if you still run into this problem

closed 2017-01-29 by SamCarlberg