WPIRoboticsProjects/GRIP #544: Java/GRIP throws SIGSEGV with jpeg_fill_bit_buffer in libopencv_imgcodecs.so.3.0 on Raspberry Pi 2

ryannazaretian, 2016-03-15:
Commented on a closed issue two weeks ago, not sure if anyone was going to respond...

Tried with both GRIP 1.2 and 1.3. (SIGSEGV with jpeg_fill_bit_buffer in libopencv_imgcodecs.so.3.0)

[hs_err_pid958.log.txt](https://github.com/WPIRoboticsProjects/GRIP/files/174470/hs_err_pid958.log.txt)

[project.grip.txt](https://github.com/WPIRoboticsProjects/GRIP/files/174471/project.grip.txt)

Here's a link to the setup script on how I setup the Raspberry Pi environment.
https://github.com/GarnetSquardon4901/rpi-vision-processing

EdWard680, 2016-03-16:
Is it happening for all projects or only when you run a certain operation. Whenever I try to run your project the current version of the UI just spams endless FrameGrabber related exceptions

ryannazaretian, 2016-03-17:
It was happening for two different projects. One was a project we started back in January. Had the issue and decided to rewrite it, but still the same issue.

rlee287, 2016-03-18:
Have you set the image size to 320x240 with 15 fps yet? If you haven't done so, I would recommend trying that to see if it helps.

ryannazaretian, 2016-03-21:
We do. Here's our mjpg_streamer script: 
`env LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH mjpg_streamer -o "output_http.so -w ./www -p 1180" -i "input_raspicam.so -fps 15 -x 320 -y 240" &`

Regardless, GRIP should lag if I gave it too much data to process, not crash.

ZeusChops, 2016-03-24:
Looking into the issue - there may be a connection between the heap size and this error being thrown.

closed 2017-01-08 by SamCarlberg