WPIRoboticsProjects/GRIP #550: GRIP not compiling using "gradlew.bat :ui:run --stacktrace"

ZeusChops, 2016-03-23:
Steps to reproduce my issue:
-GitHub Shell "git clone https://github.com/WPIRoboticsProjects/GRIP"
-close GitHub shell, open command prompt
-chdir to the GRIP project
-type "gradlew.bat :ui:run --stacktrace" and I got this: http://imgur.com/If6tEd0

OS: Windows 10 Home 64-bit
GitHub Desktop installed
Current path variable: https://goo.gl/xylZpe

ThomasJClark, 2016-03-23:
- Make sure you have Oracle JDK 8 installed
- JAVA_HOME should be set to something roughly like `C:\Program Files\Java\jdk_1.8.0_65\`

ZeusChops, 2016-03-23:
I have Java 8u74 installed, my main programming language is Java and the robotics team uses Java - every new laptop image follows a Java installation. "java -version" in cmd responds with:

java version "1.8.0_74"
Java (TM) SE Runtime Environment (build 1.8.0_74-b02)
Java HotSpot(TM) 64-Bit Server VM (build 25.74-b02, mixed mode)

Looking into JAVA_HOME settings - JAVA_HOME variable didn't exist, so I added it manually. Creating JAVA_HOME manually fixed the issue.

Steps:
-Right click in the bottom left corner
-Click on System -> Advanced System Settings -> Environment Variables -> New
-Variable name: JAVA_HOME
-Variable value: C:\Program Files\Java\jdk1.8.0_74

Thanks!

closed 2016-03-23 by ThomasJClark