WPIRoboticsProjects/GRIP #554: Filter Contours

mtmustski, 2016-03-27:
The previous version of Grip, I created a pipeline that displayed the target for the 2016 FRC competition.  It took the picture, ran it through the HSL Threshold, then found the counters, and filtered them.  However, I updated the program today in hopes of being able to find a shape instead of contours, (could not find such a feature).  When I ran the pipeline with the file from the last update, the filter contours found 0 contours, and displayed a black screen.  I am also using the same image as a base point.  I am using the same pipeline, so I am confused on why it no longer works in the new update.  Any support would be greatly appreciated.

SamCarlberg, 2016-03-27:
Sounds like a duplicate of #538

JLLeitschuh, 2016-04-11:
This should be resolved in the newest release.

closed 2016-04-11 by JLLeitschuh