WPIRoboticsProjects/GRIP #555: Building GRIP requires internet connection

rlee287, 2016-03-28:
In Eclipse, building GRIP requires an active internet connection. This may be an issue during FIRST competitions when internet access is not allowed in the pits. I will post the resulting stack trace later when I have a chance.

JLLeitschuh, 2016-03-29:
Yea, not much we can do about that. Unless @JacisNonsense knows how to cache a build script.

JacisNonsense, 2016-03-29:
Run the gradle task with the `--offline` flag. You must have run the gradle script at least once online to gather the dependencies, but after that you can run `./gradlew <task> --offline` and it will used the cached resources

JLLeitschuh, 2016-03-29:
That should really be added to the readme.

JLLeitschuh, 2016-03-30:
Closed by 2793ffa

closed 2016-03-30 by JLLeitschuh

rlee287, 2016-03-30:
@JacisNonsense Still getting errors when building offline:
```
[sts] -----------------------------------------------------
[sts] Starting Gradle build for the following tasks: 
[sts]      :ui:run
[sts] -----------------------------------------------------
:buildSrc:compileJava UP-TO-DATE
:buildSrc:extractSources UP-TO-DATE
:buildSrc:compileGroovy UP-TO-DATE
:buildSrc:processResources UP-TO-DATE
:buildSrc:classes UP-TO-DATE
:buildSrc:jar UP-TO-DATE
:buildSrc:assemble UP-TO-DATE
:buildSrc:compileTestJava UP-TO-DATE
:buildSrc:compileTestGroovy UP-TO-DATE
:buildSrc:processTestResources UP-TO-DATE
:buildSrc:testClasses UP-TO-DATE
:buildSrc:test UP-TO-DATE
:buildSrc:check UP-TO-DATE
:buildSrc:build UP-TO-DATE

FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\Ryan\Documents\GitHub\Robotics\GRIP\build.gradle' line: 278

* What went wrong:
A problem occurred evaluating root project 'GRIP'.
> Could not read script 'http://dl.bintray.com/shemnon/javafx-gradle/8.1.1/javafx.plugin'.
   > dl.bintray.com

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 3.051 secs
[sts] Build failed
org.gradle.tooling.BuildException: Could not execute build using Gradle distribution 'https://services.gradle.org/distributions/gradle-2.10-bin.zip'.
	at org.gradle.tooling.internal.consumer.ResultHandlerAdapter.onFailure(ResultHandlerAdapter.java:57)
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:57)
	at org.gradle.internal.concurrent.DefaultExecutorFactory$StoppableExecutorImpl$1.run(DefaultExecutorFactory.java:64)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
	at org.gradle.tooling.internal.consumer.BlockingResultHandler.getResult(BlockingResultHandler.java:46)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher.run(DefaultBuildLauncher.java:71)
	at org.springsource.ide.eclipse.gradle.core.TaskUtil.execute(TaskUtil.java:117)
	at org.springsource.ide.eclipse.gradle.core.launch.GradleProcess$1.doit(GradleProcess.java:92)
	at org.springsource.ide.eclipse.gradle.core.util.GradleRunnable$1.run(GradleRunnable.java:53)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
Caused by: org.gradle.internal.exceptions.LocationAwareException: Build file 'C:\Users\Ryan\Documents\GitHub\Robotics\GRIP\build.gradle' line: 278
A problem occurred evaluating root project 'GRIP'.
	at org.gradle.initialization.DefaultExceptionAnalyser.transform(DefaultExceptionAnalyser.java:74)
	at org.gradle.initialization.MultipleBuildFailuresExceptionAnalyser.transform(MultipleBuildFailuresExceptionAnalyser.java:47)
	at org.gradle.initialization.StackTraceSanitizingExceptionAnalyser.transform(StackTraceSanitizingExceptionAnalyser.java:30)
	at org.gradle.initialization.DefaultGradleLauncher$1.create(DefaultGradleLauncher.java:101)
	at org.gradle.initialization.DefaultGradleLauncher$1.create(DefaultGradleLauncher.java:93)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:90)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:62)
	at org.gradle.initialization.DefaultGradleLauncher.doBuild(DefaultGradleLauncher.java:93)
	at org.gradle.initialization.DefaultGradleLauncher.run(DefaultGradleLauncher.java:82)
	at org.gradle.launcher.exec.InProcessBuildActionExecuter$DefaultBuildController.run(InProcessBuildActionExecuter.java:94)
	at org.gradle.tooling.internal.provider.runner.BuildModelActionRunner.run(BuildModelActionRunner.java:46)
	at org.gradle.launcher.exec.ChainingBuildActionRunner.run(ChainingBuildActionRunner.java:35)
	at org.gradle.tooling.internal.provider.runner.SubscribableBuildActionRunner.run(SubscribableBuildActionRunner.java:58)
	at org.gradle.launcher.exec.ChainingBuildActionRunner.run(ChainingBuildActionRunner.java:35)
	at org.gradle.launcher.exec.InProcessBuildActionExecuter.execute(InProcessBuildActionExecuter.java:43)
	at org.gradle.launcher.exec.InProcessBuildActionExecuter.execute(InProcessBuildActionExecuter.java:28)
	at org.gradle.launcher.exec.ContinuousBuildActionExecuter.execute(ContinuousBuildActionExecuter.java:78)
[... 60 lines cut ...]
	at org.gradle.api.internal.plugins.DefaultObjectConfigurationAction.access$000(DefaultObjectConfigurationAction.java:35)
	at org.gradle.api.internal.plugins.DefaultObjectConfigurationAction$1.run(DefaultObjectConfigurationAction.java:61)
	at org.gradle.api.internal.plugins.DefaultObjectConfigurationAction.execute(DefaultObjectConfigurationAction.java:135)
	at org.gradle.api.internal.project.AbstractPluginAware.apply(AbstractPluginAware.java:46)
	at build_2y67lmcl31fhrpdqpov240axa$_run_closure7.doCall(C:\Users\Ryan\Documents\GitHub\Robotics\GRIP\build.gradle:278)
	at org.gradle.api.internal.ClosureBackedAction.execute(ClosureBackedAction.java:67)
	at org.gradle.util.ConfigureUtil.configure(ConfigureUtil.java:130)
	at org.gradle.util.ConfigureUtil.configure(ConfigureUtil.java:91)
	at org.gradle.api.internal.project.AbstractProject.project(AbstractProject.java:842)
	at org.gradle.api.internal.BeanDynamicObject$MetaClassAdapter.invokeMethod(BeanDynamicObject.java:236)
	at org.gradle.api.internal.BeanDynamicObject.invokeMethod(BeanDynamicObject.java:130)
[... cut ...]
	at build_2y67lmcl31fhrpdqpov240axa.run(C:\Users\Ryan\Documents\GitHub\Robotics\GRIP\build.gradle:273)
	at org.gradle.groovy.scripts.internal.DefaultScriptRunnerFactory$ScriptRunnerImpl.run(DefaultScriptRunnerFactory.java:91)
	... 60 more
Caused by: java.net.UnknownHostException: dl.bintray.com
	at org.gradle.internal.resource.UriResource.getInputStream(UriResource.java:100)
	at org.gradle.internal.resource.UriResource.getText(UriResource.java:69)
	... 82 more
[sts] Time taken: 0 min, 5 sec
[sts] -----------------------------------------------------
```
This is my configuration:
[image]

rlee287, 2016-03-30:
For the record, here were the old stack traces I got in case this may help:

[error1_stacktrace.txt](https://github.com/WPIRoboticsProjects/GRIP/files/196074/error1_stacktrace.txt)