WPIRoboticsProjects/GRIP #562: Error in Java code (commit 2793ffa)

rlee287, 2016-04-03:
```java
    @AutoValue
    public static abstract class Contour {
        static Contour create(double area, double centerX, double centerY, double width, double height, double solidity) {
            return new AutoValue_ContoursReport_Contour(area, centerX, centerY, width, height, solidity);
        }```

AutoValue_ContoursReport_Contour cannot be resolved to a type	ContoursReport.java	/core/src/main/java/edu/wpi/grip/core/operations/composite	line 56	Java Problem

This is probably Eclipse bugging out, but just thought I would point this out.

SamCarlberg, 2016-04-03:
Enable annotation processing in Eclipse

AustinShalit, 2016-04-23:
@rlee287 Has this issue been solved? If so, please close this issue. If not, please let us know how we can help.

rlee287, 2016-04-23:
Never got around to trying SamCarlberg's advice. I don't know what \*.jar to use as an annotation processor, but looks like our team won't be using GRIP anymore, anyway :disappointed: . I will simply close the issue for now and reopen it if our team decides to use GRIP for next year.

closed 2016-04-23 by rlee287

JLLeitschuh, 2016-04-26:
If you have feedback regarding why your team chose not to utilize GRIP I'd love any feedback you may have.

rlee287, 2016-04-27:
@JLLeitschuh This is not because of any flaw in the GRIP program, but rather because my team switched to a defensive strategy and chose not to utilize vision anymore.

JLLeitschuh, 2016-04-27:
Thats unfortunate. Hopefully you'll be back next year.

GRIP will still be here!