WPIRoboticsProjects/GRIP #564: Image processing after contours filter

Qormix, 2016-04-04:
I'm using a line report for some of our image processing but I'm unable to do this after doing a contours filter

JLLeitschuh, 2016-04-04:
#538

Qormix, 2016-04-04:
Sorry let me clarify. I would like to be able to do a filter contours then a find lines. Im unable to do so because filter contours returns contours while find lines requires an image

JLLeitschuh, 2016-04-05:
You can do so by plugging in the previous image.
Do you mean like this?
[image]

Currently, the output that you preview as a "Contours" is render only. Those lines are drawn onto the image when previewed.

Qormix, 2016-04-05:
Thats currently what we are doing. But I would like to be able find lines after filtering the contours. The only solution I having is writting software to check if the lines fit inside the bounds of a contour.

AustinShalit, 2016-08-10:
@Qormix What do you plan to do with the lines after you have them?