WPIRoboticsProjects/GRIP #566: Cannot launch GRIP through .deb package - Linux

ZeusChops, 2016-04-12:
GRIP runs fine if started via gradlew, however running GRIP through Linux there is an error thrown:

`Apr 11, 2016 8:43:53 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Apr 11, 2016 8:43:53 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.4.0
platform: /Linux/amd64/
terminate called after thowing an instance of 'std::system_error
   what():  Enable multithreading to use std::thread: Operation not permitted
Aborted`

I can't seem to find kernel logs dumped anywhere on the OS either other than GRIP.log which returns "FINE:" up until the system begins to "load singletons" which says 6ms then no logs after that.

[GRIP.log](https://drive.google.com/open?id=0B86Py5WkgumYTmFNeFJBOUNGbUU)

SamCarlberg, 2016-07-01:
@ZeusChops Why did you close this?

ZeusChops, 2016-07-01:
I no longer have the log file, and the operation didn't come back on the newer updates. I can certainly try again and see whether or not we need to reopen the issue, but if the issue isn't relevant to the new updates then it should be close - or else I'm off-base and should just leave it open :)

SamCarlberg, 2016-07-01:
It looks like the pthread issue with ntcore, I don't remember what the exact fix was. Try it again and see if it still fails.

ZeusChops, 2016-07-01:
I couldn't even get the .deb file to launch properly this time. It appears as if it may have gotten worse.

[Here's a link](https://drive.google.com/open?id=0B86Py5WkgumYeG05NGlLQmMxaHM), noting that I need to keep the files this time.

SamCarlberg, 2016-07-01:
Can you upload the raw text to a gist? [gist.guthub.com](http://gist.github.com)

ZeusChops, 2016-07-01:
The link was updated to be a gist, [here it is again](https://gist.github.com/ZeusChops/aefbc8f6a34334773d3ee8769ca39b3d) (in case Github is being weird).

ThomasJClark, 2016-07-01:
Not sure if this has changed, but the "fix" was to use a locally built copy of networktables instead of the one from the maven artifact when building Linux releases. The problem was that the pthread version that ntcore is built with doesn't work with the binaries that the JavaFX pacakger generates.

PR #648 (2016-08-02): Add Launcher Binary
closes #566 
closes #345 
closes #370 

This solves the Linux build issues!

Thank you @PeterJohnson and @FibreFoX for your work on this.
closed 2016-08-02 by JLLeitschuh