WPIRoboticsProjects/GRIP #568: Fatal error in  JRE

Dpurry23, 2016-04-13:
Mar 16, 2016 4:21:33 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Mar 16, 2016 4:21:34 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.3.0-rc1-9-g48dd139
platform: /Linux/arm/
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xac9f1e18, pid=5427, tid=3056903264
#
# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)
# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )
# Problematic frame:
# C  [libstdc++.so.6+0x6ce18]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/hs_err_pid5427.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

JLLeitschuh, 2016-04-13:
So this is running on a raspberry pi?
I think this is something for @EdWard680

EdWard680, 2016-04-13:
Details would be handy, though this has happened to me once in a while

Dpurry23, 2016-04-13:
I have since found the problem, I realized that i had been using a raspberry pi B the entire time.

AustinShalit, 2016-04-23:
@Dpurry23 Has this issue been solved?  If so, please close this request.

AustinShalit, 2016-07-30:
I am going to close this issue.  If you think there is still a problem, please reopen with a comment.

closed 2016-07-30 by AustinShalit