WPIRoboticsProjects/GRIP #574: Error starting GRIP

bobhinkle, 2016-04-24:
When trying to launch GRIP I get a "failed to launch JVM". I have 2016.1.0 updates installed and also eclipse with FRC java plugins installed. Also have java sdk and jre installed. When i start the program with the ui jar I get this error:

C:\Users\DS1\AppData\Local\GRIP\app>java -jar ui-1.4.0.jar
Apr 23, 2016 8:02:10 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Apr 23, 2016 8:02:10 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.4.0
platform: /Windows/amd64/
Apr 23, 2016 8:02:12 PM edu.wpi.grip.core.operations.network.networktables.NTMa
ager lambda$new$0
INFO: NetworkTables: TCPConnector.cpp:163 connect() to  port 1735 timed out
Exception in Application start method
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(Un
nown Source)
        at com.sun.javafx.application.LauncherImpl.launchApplication(Unknown So
rce)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)
Caused by: java.lang.RuntimeException: Exception in Application start method
        at com.sun.javafx.application.LauncherImpl.launchApplication1(Unknown S
urce)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$155
Unknown Source)
        at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.UnsatisfiedLinkError: no jniopencv_core in java.library.pa
h
        at java.lang.ClassLoader.loadLibrary(Unknown Source)
        at java.lang.Runtime.loadLibrary0(Unknown Source)
        at java.lang.System.loadLibrary(Unknown Source)
        at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:597)
        at org.bytedeco.javacpp.Loader.load(Loader.java:438)
        at org.bytedeco.javacpp.Loader.load(Loader.java:381)
        at org.bytedeco.javacpp.opencv_core.<clinit>(opencv_core.java:10)
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Unknown Source)
        at org.bytedeco.javacpp.Loader.load(Loader.java:413)
        at org.bytedeco.javacpp.Loader.load(Loader.java:381)
        at org.bytedeco.javacpp.helper.opencv_core$AbstractArray.<clinit>(openc
_core.java:109)
        at edu.wpi.grip.core.operations.opencv.MatFieldAccessor.<clinit>(MatFie
dAccessor.java:12)
        at edu.wpi.grip.core.operations.Operations.lambda$addOperations$12(Oper
tions.java:97)
        at java.util.stream.ReferencePipeline$3$1.accept(Unknown Source)
        at java.util.Iterator.forEachRemaining(Unknown Source)
        at java.util.Spliterators$IteratorSpliterator.forEachRemaining(Unknown
ource)
        at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
        at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
        at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Sou
ce)
        at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unkno
n Source)
        at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
        at java.util.stream.ReferencePipeline.forEach(Unknown Source)
        at edu.wpi.grip.core.operations.Operations.addOperations(Operations.jav
:99)
        at edu.wpi.grip.ui.Main.start(Main.java:84)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$16
(Unknown Source)
        at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$175(Unknow
 Source)
        at com.sun.javafx.application.PlatformImpl.lambda$null$173(Unknown Sour
e)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(Unknown
ource)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(Unknown Source)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$null$148(Unknown Source)
        ... 1 more
Caused by: java.lang.UnsatisfiedLinkError: C:\Users\DS1\AppData\Local\Temp\java
pp100117015399174\jniopencv_core.dll: Can't find dependent libraries
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(Unknown Source)
        at java.lang.ClassLoader.loadLibrary(Unknown Source)
        at java.lang.Runtime.load0(Unknown Source)
        at java.lang.System.load(Unknown Source)
        at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:580)
        ... 30 more
Exception running application edu.wpi.grip.ui.Main

any ideas on a fix. Also tried running GRIP version 1.3 with the same issue.

ThomasJClark, 2016-04-26:
> C:\Users\DS1\AppData\Local\GRIP\app>java -jar ui-1.4.0.jar

Run `GRIP.exe` directly instead of using `java -jar ...`.  GRIP is made up of a lot of JARS, and if you try running the main one, you won't have all of the libraries in your classpath.

bobhinkle, 2016-04-26:
When running grip.exe we get the "error starting jvm". Was trying to run from command line to try and debug the issue. Any way to start from command line so the issue can be tracked down?

rcc1323, 2016-11-14:
Can anyone help with this issue?

Thanks!

SamCarlberg, 2016-11-18:
@rcc1323 

1. What OS?
2. What version of GRIP?
3. Are you running GRIP from the app or by running the UI jar?

bobhinkle, 2016-11-18:
Tried running the latest build and build 1.4. Running on windows 7 and from
the installed version and exe.

SamCarlberg, 2016-11-18:
Seems like a bug with JavaCV not properly extracting the required libraries. Did you uninstall the old version before installing 1.4? Installing over the old version overwrites the old files, so that may have messed up the dependency jars.

I'm also unable to reproduce this on a fresh Windows 10 VM

rcc1323, 2016-11-18:
Sam any idea how we could give you a better debug log? We can also try
fresh reinstalling today.

Thanks!

-RC
Sent from Mobile

SamCarlberg, 2016-11-18:
If the stacktrace has something like this in it:

```java
Caused by: java.lang.UnsatisfiedLinkError: C:\Users\DS1\AppData\Local\Temp\java
pp100117015399174\jniopencv_core.dll: Can't find dependent libraries
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(Unknown Source)
        at java.lang.ClassLoader.loadLibrary(Unknown Source)
        at java.lang.Runtime.load0(Unknown Source)
        at java.lang.System.load(Unknown Source)
        at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:580)
        ... 30 more
```

Then it's probably caused by the same thing. If your computer is running Windows 7, then it might be specific to that OS. If it's 8/8.1 or 10, then I can rule that out because I can't reproduce this on either.