WPIRoboticsProjects/GRIP #592: Most Factory produced OpenCV operations fail to perform

PatrickPenguinTurtle, 2016-05-31:
So first I noticed that "bitwise not" takes 2 sources. I tried to fix this in the source code and found the place. After fixing it I tried running GRIP again and it gave me a value not present error. I tried several other of the Factory made operations including "bitwise and" which gave me the stack trace

WARNING: The CV bitwise_and operation did not perform correctly.
java.util.NoSuchElementException: No value present
	at java.util.Optional.get(Optional.java:135)
	at edu.wpi.grip.core.operations.templated.TwoSourceOneDestinationOperation.perform(TwoSourceOneDestinationOperation.java:48)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:151)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:184)
	at edu.wpi.grip.core.PipelineRunner.access$100(PipelineRunner.java:35)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:68)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

I had two images connected to the two inputs of the add when this occured. 
[image]

The same issue occurs with most of the Factory produced operations made inside of core/src/main/java/edu/wpi/grip/core/operations/CVOperations.java

JLLeitschuh, 2016-05-31:
Found the bug. Thanks for pointing this out.

closed 2016-06-01 by JLLeitschuh

SamCarlberg, 2016-06-01:
What's with all the asserts?

JLLeitschuh, 2016-06-01:
It will make the tests fail if the contract of the classes aren't fulfilled.

PR #593 (2016-06-01): Fix Templated Operations not running
Closes #592
commit c49a7e0a (2016-06-01): Fix Templated Operations not running (#593)

Closes #592