WPIRoboticsProjects/GRIP #594: URLDecoder.decode(String) is deprecated

AustinShalit, 2016-06-01:
ui/src/main/java/edu/wpi/grip/ui/DeployController.java uses URLDecoder.decode(String) -- a deprecated API.  Should be updated to the explicit API.

[URLDecoder (Java Platform SE 8)](https://docs.oracle.com/javase/8/docs/api/java/net/URLDecoder.html)

JLLeitschuh, 2016-06-01:
Yea, I saw that. Didn't know how to fix it.
If you want to take a crack at it go ahead.

ZeusChops, 2016-06-29:
There is now a URLDecoder.decode(String, String). Usage: URLDecoder.decode(String, "UTF-8") would work as a replacement for the current URLDecoder.decode(String).

Note: URLDecoder.decode(String, String) throws an UnsupportedEncodingException if the wrong value is put in for the second string - instead of "UTF-8" you put "UTF8" or something else - i.e. "NTFS".

PR #657 (2016-08-08): Cleanup of deprecated method usages
closes #594
closed 2016-08-08 by AustinShalit