WPIRoboticsProjects/GRIP #622: GRIP cron task to deploy

AustinShalit, 2016-07-11:
We should probably add the option to create a cron task for grip on deploy.

JLLeitschuh, 2016-07-12:
Cron task?

SamCarlberg, 2016-07-12:
Linux cron job to run GRIP, probably on boot, if it's been deployed. Besically [this](http://stackoverflow.com/questions/5034333/creating-cron-entry-on-server-using-ssh-login-within-shell-script)

SamCarlberg, 2016-10-31:
Do we still need this? I feel like the deploy mechanism is superfluous now that we have code generation

JLLeitschuh, 2016-10-31:
I think its still relevant.
Unless you can deploy code gen like you can deploy GRIP I think deployment makes sense to keep in GRIP.

SamCarlberg, 2016-10-31:
You can generate code in a robot program directory and deploy that like a normal FRC program. Or save it to another project directory and FTP it/copy on a flashdrive to a raspberry pi or jetson or kangaroo or...

JLLeitschuh, 2016-10-31:
Yea, but that doesn't include sources/network tables publishing.

SamCarlberg, 2016-10-31:
Right, they'd have to add that code themselves. Again, we should do a survey on the use of deploy vs code gen during/after the 2017 season

JLLeitschuh, 2016-10-31:
Sure.

closed 2017-01-08 by SamCarlberg