WPIRoboticsProjects/GRIP #660: ⌘Q -> Cancel still quits GRIP

AustinShalit, 2016-08-09:
Using the shortcut ⌘Q and pressing cancel on the dialog still quits GRIP when it should just close the dialog.

ZeusChops, 2016-08-15:
This happens sometimes. I've just ran GRIP ten times and tried the steps found in #543 and never hit a deadlock exception. To add to that thought, using [Command + Q] is macOS's shortcut for force quitting an application. Also, it shows the shortcut for the exit button in the drop-down as [Command + Q]..

Is there a video or something of this acting not as explained above?

SamCarlberg, 2016-08-15:
See https://bugs.openjdk.java.net/browse/JDK-8117679

We'll need to refactor the confirmation dialog stuff so it can get used in an event handler for the root stage

PR #689 (2016-12-31): Fix save dialog not appearing when closing an edited project
Closes #660
Closes #688
commit 9693260a (2016-12-31): Fix save dialog not appearing when closing an edited project (#689)

* Fix save dialog not appearing when closing an edited project

Closes #660
Closes #688
closed 2016-12-31 by SamCarlberg