WPIRoboticsProjects/GRIP #667: Windows 10 - grip not connecting to the webcam

ZeusChops, 2016-08-23:
Microsoft's update to windows 10 anniversary update may have caused issues with connecting to webcams.

Currently known fix: Try closing all applications using a webcam (if you're using Cortana's voice activation, you will likely need to reboot your PC), then relaunch grip. If the issue persists please comment on this issue, thanks.

Referencing link (why this might happen): https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/9d6a8704-764f-46df-a41c-8e9d84f7f0f3/mjpg-encoded-media-type-is-not-available-for-usbuvc-webcameras-after-windows-10-version-1607-os?forum=mediafoundationdevelopment#fc5c100a-c661-43cd-9540-bb4591e3d1fe

SamCarlberg, 2016-11-09:
This should be a wiki topic imo

ZeusChops, 2016-11-09:
Would it really be worth going through the hurdle of creating a wiki page for something related to the OS specifically? If this was not a known bug, according to Microsoft, or required a customized setup for a runtime environment, then I think that would have been a valid choice from the beginning.