WPIRoboticsProjects/GRIP #669: Cannot utilize Find Contours (v.1.5.0-rc1): operation did not perform correctly

ZeusChops, 2016-08-27:
Tested on both Mac OS X 10.11.5 and Windows 10 (via virtualbox >> Windows 10 Pro).

The only difference on the OS X version is the absolute path which is: `/Users/saudet/projects/bytedeco/javacpp-presets/opencv/cppbuild/macosx-x86_64/opencv-3.0.0/modules/imgproc/src/contours.cpp:198`

I have not yet tried this on Linux. The steps in these pipelines were: `Camera 0 input` >> `Resize Image` >> `Find Contours`

If Find Contours is not fed a connection within the pipeline, the issue is not created. The issue only occurs while a source is connected to the pipeline.

Exception (from Windows 10):
```
java.lang.RuntimeException: src\contours.cpp:198: error: (-210) [Start]FindContours supports only CV_8UC1 images when mode != CV_RETR_FLOODFILL otherwise supports CV_32SC1 images only in function cvStartFindContours
 at org.bytedeco.javacpp.opencv_imgproc.findContours(Native Method)
 at edu.wpi.grip.core.operations.composite.FindContoursOperation.perform(FindContoursOperation.java:95)
 at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:119)
 at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:198)
 at edu.wpi.grip.core.PipelineRunner.access$100(PipelineRunner.java:40)
 at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:80)
 at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
 at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
 at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
 at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
 at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
 at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
 at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
 at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
 at java.lang.Thread.run(Thread.java:745)
```

Main gist of GRIP.log:
```
Aug 27, 2016 4:13:28 PM edu.wpi.grip.core.Step runPerformIfPossible
WARNING: The Find Contours operation did not perform correctly.
java.lang.RuntimeException: /Users/saudet/projects/bytedeco/javacpp-presets/opencv/cppbuild/macosx-x86_64/opencv-3.0.0/modules/imgproc/src/contours.cpp:198: error: (-210) [Start]FindContours supports only CV_8UC1 images when mode != CV_RETR_FLOODFILL otherwise supports CV_32SC1 images only in function cvStartFindContours

	at org.bytedeco.javacpp.opencv_imgproc.findContours(Native Method)
	at edu.wpi.grip.core.operations.composite.FindContoursOperation.perform(FindContoursOperation.java:95)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:119)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:198)
	at edu.wpi.grip.core.PipelineRunner.access$100(PipelineRunner.java:40)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:80)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:189)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
```

AustinShalit, 2016-08-27:
This works as expected.  You need to pass the image through a threshold operation (HSV Threshold) first.

closed 2016-08-27 by AustinShalit

ZeusChops, 2016-08-28:
My mistake - after more research, [the reference table](https://github.com/WPIRoboticsProjects/GRIP/wiki/Operation-Reference-Table) only explains that the image should be a binary color (black/white pixels), would this also apply to other thresholds?