WPIRoboticsProjects/GRIP #670: I can No longer auto launch the Grip app and load a pipeline

cpapplefamily, 2016-09-04:
With the latest update I can no longer Auto launch the Grip app on the Kangaroo with our pipe line the shortcut is  C:\Users\apple\AppData\Local\GRIP\GRIP.exe "C:\Users\apple\Documents\FRC GRIP FILES\Test.grip"

Do I have the shortcut wrong? I did recreate the pipe line because the new version would not open the old file.

cpapplefamily, 2016-09-04:
Reading other current issues seems this maybe in line with the issue "not launching the JVM". If I delete the file test.grip from the shortcut GRIP will launch.

cpapplefamily, 2016-09-06:
I just went back to GRIP-1.4.0-x64 and the short cut with the file parameter works. Could be something with the new launch window?

JLLeitschuh, 2016-09-06:
@AustinShalit You know anything about this?

AustinShalit, 2016-09-06:
I will look into this

AustinShalit, 2016-09-06:
@cpapplefamily Thank you for finding this bug!  I will fix it tonight.  It was created in #629.

JLLeitschuh, 2016-09-06:
What's the cause?

AustinShalit, 2016-09-06:
Opening the pipeline before the list of operations has been created.  

https://github.com/WPIRoboticsProjects/GRIP/blob/64de5d988e0481b51ffd2058694d8ae54ec8398a/ui/src/main/java/edu/wpi/grip/ui/Main.java#L103-L123

JLLeitschuh, 2016-09-07:
That would do it.

PR #671 (2016-09-07): Solve ordering issue when opening a file via command line
closes #670
closed 2016-09-07 by JLLeitschuh