WPIRoboticsProjects/GRIP #684: Can't generate code

YairZiv5951, 2016-10-12:
Please describe what actions we can take to reproduce the bug you found, including the error message.

I pressed Generate code and it threw a velocity exception.

JLLeitschuh, 2016-10-12:
Can you upload a zip of your `.grip` save file as well as the exception you experienced?

MoSadie, 2016-10-27:
I'm experiencing the same issue. Here is a gist of the exception: https://gist.github.com/MoSadie/f1f5a2c7808b3330b52d6d1482cac270

Here is a link to the .grip project file: https://1drv.ms/u/s!AgQZOPyu2Ulvj6lPWN-UV1b85qMq_w

SamCarlberg, 2016-10-28:
I confirmed this happens with 1.5.0-rc2, but I'm not seeing it on the current master (54d98ea)

Can you download [the master zip](https://github.com/WPIRoboticsProjects/GRIP/archive/master.zip) and try again?

MoSadie, 2016-10-28:
That appears to have solved the problem! Thanks!

SamCarlberg, 2016-10-28:
Never mind. Just built that commit as a native app on Windows 10 and it still fails.

PR #694 (2016-10-29): Fix code generation with Windows native app
Closes #684
commit 765dd485 (2016-10-29): Fix code generation with Windows native app (#694)

Closes #684
closed 2016-10-29 by AustinShalit