WPIRoboticsProjects/GRIP #690: Can't Launch GRIP on RPI 3

JDaniel4468, 2016-10-15:
Hello everyone,
My team is trying to use GRIP on our Raspberry Pi 3 to run the vision code for our robot. Each time we deploy, we keep recieving an error that I have pasted at the bottom of this post. We've followed all of the steps at the wiki and have copied over the shared libraries. Each time we deploy, it keeps saying that it can't instantiate files related to network tables. I'm sure that we have placed the shared libraries in the right folders and have deployed the project to the right folder, so do you know why this isn't working? I'm using GRIP 1.2 on Raspbian.
Thanks,
Jonathan Daniel
Error:
/usr//bin/java -Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -jar '/home/pi/code/grip/grip.jar' '/home/pi/code/grip/project.grip'
Loading Dependency Injection Framework
Oct 15, 2016 3:58:43 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Oct 15, 2016 3:58:43 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.2.0-5-g3a658db
platform: /Linux/arm/
Oct 15, 2016 3:58:45 PM edu.wpi.grip.core.GRIPCoreModule onThreadException
SEVERE: Uncaught Exception on thread Thread[main,5,main]
com.google.inject.ProvisionException: Unable to provision, see the following errors:
1) Error injecting constructor, java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
at edu.wpi.grip.core.operations.networktables.NTManager.<init>(Unknown Source)
at edu.wpi.grip.core.operations.networktables.NTManager.class(Unknown Source)
while locating edu.wpi.grip.core.operations.networktables.NTManager
for field at edu.wpi.grip.core.Pipeline.ntManager(Unknown Source)
at edu.wpi.grip.core.Pipeline.class(Unknown Source)
while locating edu.wpi.grip.core.Pipeline
for field at edu.wpi.grip.core.serialization.Project.pipeline(Unknown Source)
at edu.wpi.grip.core.serialization.Project.class(Unknown Source)
while locating edu.wpi.grip.core.serialization.Project
for field at edu.wpi.grip.core.Main.project(Unknown Source)
while locating edu.wpi.grip.core.Main
Caused by: java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1864)
at java.lang.Runtime.loadLibrary0(Runtime.java:870)
at java.lang.System.loadLibrary(System.java:1122)
at edu.wpi.first.wpilibj.networktables.NetworkTablesJNI.<clinit>(NetworkTablesJNI.java:57)
at edu.wpi.grip.core.operations.networktables.NTManager.<init>(NTManager.java:50)
at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
at com.google.inject.internal.DefaultConstructionProxyFactory$1.newInstance(DefaultConstructionProxyFactory.java:86)
at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:105)
at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1103)
at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:145)
at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
at com.google.inject.internal.SingleFieldInjector.inject(SingleFieldInjector.java:54)
at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:114)
at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1103)
at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:145)
at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
at com.google.inject.internal.SingleFieldInjector.inject(SingleFieldInjector.java:54)
at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:114)
at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1103)
at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:145)
at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
at com.google.inject.internal.SingleFieldInjector.inject(SingleFieldInjector.java:54)
at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:114)
at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
[... 117 lines cut ...]
at com.google.inject.internal.SingleParameterInjector.getAll(SingleParameterInjector.java:62)
at com.google.inject.internal.SingleMethodInjector.inject(SingleMethodInjector.java:84)
at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:114)
at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1103)
at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:145)
at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
at com.google.inject.internal.SingleFieldInjector.inject(SingleFieldInjector.java:54)
at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:114)
at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
at com.google.inject.internal.InjectorImpl$2$1.call(InjectorImpl.java:1016)
at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1012)
at com.google.inject.internal.InjectorImpl.getInstance(InjectorImpl.java:1051)
at edu.wpi.grip.core.Main.main(Main.java:43)
3 errors
at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1025)
at com.google.inject.internal.InjectorImpl.getInstance(InjectorImpl.java:1051)
at edu.wpi.grip.core.Main.main(Main.java:43)

SamCarlberg, 2016-10-15:
In `/home/pi/.bash_profile`, add this line:

```bash
set LD_LIBRARY_PATH=/home/pi/vision/grip:$LD_LIBRARY_PATH
```

This will set the `LD_LIBRARY_PATH` environment variable, which is a path that the JVM looks at to find JNI bindings at startup. You may need to reboot after doing this.

JDaniel4468, 2016-10-15:
If that doesn't work, would you recommend something else?

SamCarlberg, 2016-10-15:
Then you'll have to use

```bash
env LD_LIBRARY_PATH=/home/pi/vision/grip:$LD_LIBRARY_PATH java -jar /home/pi/vision/grip/grip.jar /home/pi/vision/grip/project.grip
```

every time you run GRIP.

JDaniel4468, 2016-10-15:
So, does this tell the Raspberry Pi where to look for Network Tables, which is the solution?

SamCarlberg, 2016-10-15:
Basically.

The NetworkTables jar includes the JNI bindings for Windows, OSX, and Linux for the x86_64 architecture. Since the pi is running on ARM, you need to install the JNI manually and tell the JVM where to find it. Setting `LD_LIBRARY_PATH` is one of of doing it.

JDaniel4468, 2016-10-15:
OK, can the wiki for the instructions be updated to reflect this on the Pi 3?

SamCarlberg, 2016-10-15:
The second alternative is at the bottom of the [Deploying GRIP](https://github.com/WPIRoboticsProjects/GRIP/wiki/Running-GRIP-on-a-Raspberry-Pi-2#deploying-grip) section. (That's where I took it from, actually). It's not the best solution, though.

JDaniel4468, 2016-10-16:
OK. Also, I had one more question. Does it matter what I put for the JVM options? Currently, the default is -Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError

SamCarlberg, 2016-10-17:
Those JVM options are only used because the RoboRIO is so memory-limited (it only has 256MB)

`-Xmx50m` limits the GRIP JVM to 50MB of memory use. Otherwise, it could use up all the available memory and crash (or cause other applications -- like your robot program! -- to crash). The Pi 3 has 1GB of RAM, so you can increase the limit.

`-XX:-OmitStackTraceInFastThrow` makes the full stack traces get filled out. Otherwise you'll just see the error message and the line that threw it, without any of the stack information.

`-XX:+HeapDumpOnOutOfMemoryError` does a heap dump when the JVM runs out of memory so we can see what was taking up so much memory.

JDaniel4468, 2016-10-18:
OK. I just tried your solution and it resolved my earlier problem. However, now it's saying that it cant create a JVM and commands like HSV Filtering and Dilate are failing due to not having enough inputs. Would it be possible to just run GRIP on a driver station and plug the camera into a USB port on the RIO?

SamCarlberg, 2016-10-18:
Please post the logs

JDaniel4468, 2016-10-19:
I don't have the log of the error message because I tried deploying to the RIO afterwards, which led to the error that caused me to create this thread. But it was saying something like it couldn't run HSV Filtering and CV:Dilate due to not having a certain number of inputs, even though I connected them on the pipeline. It also said that it failed to create the Java Virtual Machine.

SamCarlberg, 2016-10-19:
The inputs thing seems to happen when an image source (webcam, IP camera, etc) throws an exception when getting initialized. There's probably an issue with connecting to your webcam. Is it plugged into the pi?

The JVM can't be initialized if there's not enough free memory.

JDaniel4468, 2016-10-19:
For the Pi test, the webcam was plugged directly into the USB Port. It was running mjpg-streamer. However, why would the JVM not be initialized from a lack of memory? Nothing else is running on the Pi, should I up the limit in the JVM Deploy Settings?

closed 2017-01-08 by SamCarlberg