WPIRoboticsProjects/GRIP #695: Ubuntu 16.10: Cancel on error quits GRIP UI

ZeusChops, 2016-10-31:
NullPointerException when removing objects from pipeline. Two case scenarios:

Case 1:
   -Setup pipeline, connect some steps
   -Start deleting steps from pipeline that are used by other steps (slowly)
   -Receive an error about a NullPointerException from the pipeline
   -Hit cancel on the message, then the UI quits out unexpectedly

Case 2:
   -Start up Ubuntu without a camera connected (or unplug all cameras from ubuntu before launching GRIP)
   -Add camera to GRIP without physical connection
   -Verify that the Camera is 'connected' using the flashing red 'X' next to the step in the sources pane
   -Remove camera from the pipeline, UI crashes (there might be another UI error window that displays, but I have not yet confirmed that)

Device: mid-2015 MacBook Pro
Type: Virtual Machine, guest
OS: Ubuntu 16.10 x64

VirtualBox configuration:
RAM: 8GB
CPU: i7 2.2GHz (quad-core, not default)
PAX/NE enabled (not default)
VRAM: 128MB (not default)
1 Screen (default)
HiDPI (2.0)
Network: NAT (default)
Audio: ICH AC97 (not default)
Execution Cap: 100% (default)
[VirtualBox Hard Drive download](https://drive.google.com/open?id=0B86Py5WkgumYVjUxSTM0QXlNTGc)

SamCarlberg, 2016-10-31:
Canceling the error message is supposed to do that, since our only shows up on fatal exceptions. The other stuff is pretty weird, can you take a video of it?

SamCarlberg, 2016-10-31:
I think this is related. I opened an instance of GRIP and added a webcam source, then opened another instance and added and removed a webcam source. Got this stacktrace when removing the webcam source

## Stack Trace:
```java
java.lang.IllegalStateException: Expected the service to be TERMINATED, but the service has FAILED
	at com.google.common.util.concurrent.AbstractService.checkCurrentState(AbstractService.java:310)
	at com.google.common.util.concurrent.AbstractService.awaitTerminated(AbstractService.java:280)
	at com.google.common.util.concurrent.AbstractExecutionThreadService.awaitTerminated(AbstractExecutionThreadService.java:217)
	at edu.wpi.grip.core.util.service.AutoRestartingService.stopAndAwait(AutoRestartingService.java:161)
	at edu.wpi.grip.core.sources.CameraSource.stopAndAwait(CameraSource.java:288)
	at edu.wpi.grip.core.sources.CameraSource.onSourceRemovedEvent(CameraSource.java:341)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:95)
	at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:154)
	at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:80)
	at com.google.common.util.concurrent.MoreExecutors$DirectExecutor.execute(MoreExecutors.java:456)
	at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:76)
	at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:119)
	at com.google.common.eventbus.EventBus.post(EventBus.java:215)
	at edu.wpi.grip.ui.pipeline.source.SourceController.delete(SourceController.java:100)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor1.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1771)
	at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1657)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$ClickGenerator.postProcess(Scene.java:3470)
	at javafx.scene.Scene$ClickGenerator.access$8100(Scene.java:3398)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3766)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:380)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:294)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:416)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:415)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
[... cut ...]
	... 1 more
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: cvCreateCameraCapture() Error: Could not create camera capture.
	at org.bytedeco.javacv.OpenCVFrameGrabber.start(OpenCVFrameGrabber.java:179)
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:52)
	... 3 more

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | 1.5.0-rc2-3-g54d98ea
java.version | 1.8.0_102
javafx.version | 8.0.102
os.name | Linux
os.version | 4.4.0-45-generic
os.arch | amd64

PR #696 (2016-11-08): Fix #695
Closes #695 

695 brought up two problems:

1. Removing a step with connections from the pipeline would cause an exception to be thrown; and  
2. Removing a failing camera source would also cause an exception (unrelated to point 1)
commit 0af249af (2016-11-08): Fix #695 (#696)

* Fix #695
closed 2016-11-08 by SamCarlberg