WPIRoboticsProjects/GRIP #703: Saving grip file causes exception

nollchr, 2016-11-05:
Trying to save a grip file throws the following exception:
Thread[JavaFX Application Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1774)
	at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1657)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.control.MenuItem.fire(MenuItem.java:462)
	at com.sun.javafx.scene.control.skin.ContextMenuContent$MenuItemContainer.doSelect(ContextMenuContent.java:1405)
	at com.sun.javafx.scene.control.skin.ContextMenuContent$MenuItemContainer.lambda$createChildren$343(ContextMenuContent.java:1358)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:380)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:294)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:416)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:415)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor1.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1771)
	... 43 more
Caused by: com.thoughtworks.xstream.converters.ConversionException: Failed calling method
---- Debugging information ----
message             : Failed calling method
cause-exception     : com.thoughtworks.xstream.core.TreeMarshaller$CircularReferenceException
cause-message       : Recursive reference to parent object
method              : java.util.concurrent.CopyOnWriteArrayList.writeObject()
-------------------------------
	at com.thoughtworks.xstream.core.util.SerializationMembers.callWriteObject(SerializationMembers.java:151)
	at com.thoughtworks.xstream.converters.reflection.SerializableConverter.doMarshal(SerializableConverter.java:257)
[... 60 lines cut ...]
	at com.thoughtworks.xstream.core.TreeMarshaller.convert(TreeMarshaller.java:70)
	at com.thoughtworks.xstream.core.TreeMarshaller.convertAnother(TreeMarshaller.java:58)
	at com.thoughtworks.xstream.core.TreeMarshaller.convertAnother(TreeMarshaller.java:43)
	at com.thoughtworks.xstream.converters.reflection.SerializableConverter$1.writeToStream(SerializableConverter.java:139)
	at com.thoughtworks.xstream.core.util.CustomObjectOutputStream.writeObjectOverride(CustomObjectOutputStream.java:87)
	at java.io.ObjectOutputStream.writeObject(ObjectOutputStream.java:344)
	at java.util.concurrent.CopyOnWriteArrayList.writeObject(CopyOnWriteArrayList.java:972)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.thoughtworks.xstream.core.util.SerializationMembers.callWriteObject(SerializationMembers.java:147)
	... 100 more

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | 1.5.0-rc3
java.version | 1.8.0_101
javafx.version | 8.0.101
os.name | Windows 10
os.version | 10.0
os.arch | amd64

SamCarlberg, 2016-11-05:
What did you have in the pipeline?

nollchr, 2016-11-05:
Hi Sam

All I had was a simple image source block that was reading a png file as the source

nollchr, 2016-11-05:
Image source block reading a png file crashes

First crash had image source piped to   Hsv piped to detect contour pipe to publish contour.  When that crashed I just tried the simple image source only and that crashed too.

SamCarlberg, 2016-11-05:
I'm not seeing this so far on Linux or my Windows 10 VM. I'll take a closer look over the weekend

SamCarlberg, 2016-11-08:
I can't reproduce this on any platform. @nollchr, can you reinstall GRIP and see if this still happens?

nollchr, 2016-11-08:
Will do.  I'll also check what the path was for the save file.  I know it was an odd path in the user documents area and might have had spaces or other characters in it.  I'll try saving to root of c drive to see if it happens.

SamCarlberg, 2016-11-18:
@nollchr Any updates?

nollchr, 2016-11-18:
Hi Sam.  Reinstalled and still have the problem.  Also tried 32-bit version and had same problem.  I've been working with it on my MacBook since with no issues.  I'll try to get back on the win10 laptop and see if I can find any hints as to what is causing this.  Assuming nobody else has experienced this?

SamCarlberg, 2016-11-18:
Did you uninstall the previous version, or just re-run the installer? If you did that and it's still happening, what steps do you have in the pipeline? A screenshot would be best

nollchr, 2016-11-18:
Uninstalled and reran installer.  Let me grab the laptop and get you some data.

nollchr, 2016-11-19:
Path to user area contains a space in it..not sure if that is the issue
C:\Users\GEH-Magnet School 1\DocumentsThe save did manage to create a .grip file, which crashes grip when it gets loaded
Including .grip file plus source image (but pipeline contents don't really seem to matter...empty pipeline save crashes as well

 

   
<edu.wpi.grip.core.Pipeline>
  <sources>
    <edu.wpi.grip.core.sources.ImageFileSource>
      <property name="path" value="C:\Users\GEH-Magnet School 1\Pictures\rectangle.png"/>
    </edu.wpi.grip.core.sources.ImageFileSource>
  </sources>
  <steps>
    <edu.wpi.grip.core.Step name="HSV Threshold">
      <edu.wpi.grip.core.sockets.InputSocketImpl step="0" socket="0"/>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="0" socket="1">
        <value>
          <double>45.32374100719424</double>
          <double>126.96969696969698</double>
        </value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="0" socket="2">
        <value>
          <double>45.86330935251798</double>
          <double>186.31313131313132</double>
        </value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="0" socket="3">
        <value>
          <double>0.0</double>
          <double>255.0</double>
        </value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.OutputSocketImpl step="0" socket="0" previewed="false"/>
    </edu.wpi.grip.core.Step>
    <edu.wpi.grip.core.Step name="Find Contours">
      <edu.wpi.grip.core.sockets.InputSocketImpl step="1" socket="0"/>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="1" socket="1">
        <value>false</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.OutputSocketImpl step="1" socket="0" previewed="true"/>
    </edu.wpi.grip.core.Step>
    <edu.wpi.grip.core.Step name="Filter Contours">
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="0"/>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="1">
        <value>300.0</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="2">
        <value>0</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="3">
        <value>0</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="4">
        <value>1000</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="5">
        <value>0</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="6">
        <value>1000</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="7">
        <value>
          <int>0</int>
          <int>100</int>
        </value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="8">
        <value>1000000</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="9">
        <value>0</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="10">
        <value>0</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="11">
        <value>1000</value>
      </edu.wpi.grip.core.sockets.InputSocketImpl>
      <edu.wpi.grip.core.sockets.OutputSocketImpl step="2" socket="0" previewed="true"/>
    </edu.wpi.grip.core.Step>
  </steps>
  <connections>
    <edu.wpi.grip.core.Connection>
      <edu.wpi.grip.core.sockets.OutputSocketImpl source="0" socket="0" previewed="false"/>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="0" socket="0"/>
    </edu.wpi.grip.core.Connection>
    <edu.wpi.grip.core.Connection>
      <edu.wpi.grip.core.sockets.OutputSocketImpl step="0" socket="0" previewed="false"/>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="1" socket="0"/>
    </edu.wpi.grip.core.Connection>
    <edu.wpi.grip.core.Connection>
      <edu.wpi.grip.core.sockets.OutputSocketImpl step="1" socket="0" previewed="true"/>
      <edu.wpi.grip.core.sockets.InputSocketImpl step="2" socket="0"/>
    </edu.wpi.grip.core.Connection>
  </connections>
  <eventBus>
    <identifier>default</identifier>
    <executor class="com.google.common.util.concurrent.MoreExecutors$DirectExecutor">INSTANCE</executor>
    <exceptionHandler class="null"/>
    <subscribers>
      <subscribers class="concurrent-hash-map">
        <entry>
          <java-class>edu.wpi.grip.core.events.SourceRemovedEvent</java-class>
          <java.util.concurrent.CopyOnWriteArraySet>
            <al serialization="custom">
                <default/>
                <int>5</int>
                <com.google.common.eventbus.Subscriber_-SynchronizedSubscriber>
                  <bus

SamCarlberg, 2016-11-19:
Spaces in the path should be fine. Looks like something is getting saved when it shouldn't be. Can you download the source [here](https://github.com/WPIRoboticsProjects/GRIP/archive/v1.5.0-rc3.zip) and run `gradlew :ui:run` and `gradlew jfxRun` to see if this is a problem with the native app or with our code.

nollchr, 2016-11-19:
gradlew :ui:run fails same way with following dump and gradlew jfxRun hangs (Not Responding) when I save the pipelineNov 18, 2016 9:04:22 PM edu.wpi.grip.core.util.service.LoggingListener starting
INFO: [PipelineRunner] Starting
Nov 18, 2016 9:06:01 PM edu.wpi.grip.core.GripCoreModule onThreadException
SEVERE: Uncaught Exception on thread Thread[JavaFX Application Thread,5,main]
java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
        at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1774)
        at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1657)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
        at javafx.event.Event.fireEvent(Event.java:198)
        at javafx.scene.control.MenuItem.fire(MenuItem.java:462)
        at com.sun.javafx.scene.control.skin.ContextMenuContent$MenuItemContainer.doSelect(ContextMenuContent.java:1405)
        at com.sun.javafx.scene.control.skin.ContextMenuContent$MenuItemContainer.lambda$createChildren$343(ContextMenuContent.java:1358)
        at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
        at javafx.event.Event.fireEvent(Event.java:198)
        at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
        at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
        at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
        at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
        at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
        at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
        at com.sun.glass.ui.View.notifyMouse(View.java:937)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
        at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
        at sun.reflect.GeneratedMethodAccessor3.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
        at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1771)
        ... 43 more
Caused by: com.thoughtworks.xstream.converters.ConversionException: Failed calling method
---- Debugging information ----
message             : Failed calling method
[... cut ...]
        at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
        at java.lang.Thread.run(Thread.java:745):ui:run FAILEDFAILURE: Build failed with an exception.* What went wrong:
Execution failed for task ':ui:run'.
> Process 'command 'C:\Program Files\Java\jdk1.8.0_73\bin\java.exe'' finished with non-zero exit value 1* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.BUILD FAILEDTotal time: 3 mins 42.555 secsC:\Users\GEH-Magnet School 1\Documents\GRIP-1.5.0-rc3>

SamCarlberg, 2016-11-28:
@nollchr I'm working on a potential fix on a branch on my fork. Can you download [this archive](https://github.com/SamCarlberg/GRIP/archive/fix/703.zip) and run `gradlew :ui:run` in the command line?

nollchr, 2016-11-28:
I'll try it tonight when I get home

nollchr, 2016-11-29:
Sam,Your fix worked!Thanks,Chris.

PR #722 (2016-11-29): Make Pipeline.eventBus transient
Closes #703
closed 2016-11-29 by JLLeitschuh