WPIRoboticsProjects/GRIP #704: Grip rc3 crashes with out much effort

kglasnapp, 2016-11-07:
Grip rc3 crashes with out much effort. I just loaded a file and it crashed. This has happened with many files. Once the crash happens the only thing you can do is kill grip.

Thread[Project Open Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.IllegalStateException: Not on FX application thread; currentThread = Project Open Thread
	at com.sun.javafx.tk.Toolkit.checkFxUserThread(Toolkit.java:236)
	at com.sun.javafx.tk.quantum.QuantumToolkit.checkFxUserThread(QuantumToolkit.java:423)
	at javafx.scene.Parent$2.onProposedChange(Parent.java:367)
	at com.sun.javafx.collections.VetoableListDecorator.setAll(VetoableListDecorator.java:113)
	at com.sun.javafx.collections.VetoableListDecorator.setAll(VetoableListDecorator.java:108)
	at com.sun.javafx.scene.control.skin.LabeledSkinBase.updateChildren(LabeledSkinBase.java:575)
	at com.sun.javafx.scene.control.skin.LabeledSkinBase.handleControlPropertyChanged(LabeledSkinBase.java:204)
	at com.sun.javafx.scene.control.skin.LabelSkin.handleControlPropertyChanged(LabelSkin.java:49)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase.lambda$registerChangeListener$61(BehaviorSkinBase.java:197)
	at com.sun.javafx.scene.control.MultiplePropertyChangeListenerHandler$1.changed(MultiplePropertyChangeListenerHandler.java:55)
	at javafx.beans.value.WeakChangeListener.changed(WeakChangeListener.java:89)
	at com.sun.javafx.binding.ExpressionHelper$SingleChange.fireValueChangedEvent(ExpressionHelper.java:182)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.StringPropertyBase.fireValueChangedEvent(StringPropertyBase.java:103)
	at javafx.beans.property.StringPropertyBase.markInvalid(StringPropertyBase.java:110)
	at javafx.beans.property.StringPropertyBase.set(StringPropertyBase.java:144)
	at javafx.beans.property.StringPropertyBase.set(StringPropertyBase.java:49)
	at javafx.beans.property.StringProperty.setValue(StringProperty.java:65)
	at javafx.scene.control.Labeled.setText(Labeled.java:145)
	at edu.wpi.grip.ui.pipeline.input.RangeInputSocketController.lambda$initialize$2(RangeInputSocketController.java:91)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:349)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.DoublePropertyBase.fireValueChangedEvent(DoublePropertyBase.java:106)
	at javafx.beans.property.DoublePropertyBase.markInvalid(DoublePropertyBase.java:113)
	at javafx.beans.property.DoublePropertyBase.set(DoublePropertyBase.java:147)
	at org.controlsfx.control.RangeSlider.setLowValue(RangeSlider.java:227)
	at edu.wpi.grip.ui.pipeline.input.RangeInputSocketController.updateSliderValue(RangeInputSocketController.java:108)
	at sun.reflect.GeneratedMethodAccessor47.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:95)
	at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:154)
	at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:80)
	at com.google.common.util.concurrent.MoreExecutors$DirectExecutor.execute(MoreExecutors.java:456)
	at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:76)
	at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:119)
	at com.google.common.eventbus.EventBus.post(EventBus.java:215)
	at edu.wpi.grip.core.sockets.SocketImpl.setValueOptional(SocketImpl.java:62)
	at edu.wpi.grip.core.sockets.Socket.setValue(Socket.java:56)
	at edu.wpi.grip.core.serialization.SocketConverter.unmarshal(SocketConverter.java:141)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at edu.wpi.grip.core.serialization.StepConverter.unmarshal(StepConverter.java:74)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:73)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:503)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:429)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:281)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1230)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1214)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1085)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:115)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:94)
	at edu.wpi.grip.ui.MainWindowController.lambda$openProject$2(MainWindowController.java:171)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | 1.5.0-rc3
java.version | 1.8.0_101
javafx.version | 8.0.101
os.name | Windows 10
os.version | 10.0
os.arch | amd64

[Home-11-7a.grip.txt](https://github.com/WPIRoboticsProjects/GRIP/files/575407/Home-11-7a.grip.txt)

SamCarlberg, 2016-11-07:
This is a known issue that will be fixed before the full release

SamCarlberg, 2016-11-07:
Closed with #705

closed 2016-11-07 by SamCarlberg

kglasnapp, 2016-11-07:
Sam – the reported problem is resolve resolved with a build of the new version – Thanks.

 

I have found other crashes, would you like me to report them?

Keith

 

Closed #704 <https://github.com/WPIRoboticsProjects/GRIP/issues/704> .

SamCarlberg, 2016-11-07:
Sure! Just make sure it's not a duplicate