WPIRoboticsProjects/GRIP #712: OS X: Removing a webcam source deadlocks

SamCarlberg, 2016-11-10:
Reported by @AustinShalit 

I don't have an apple computer so I can't tell if it's a general OS X thing or if it's specific to Austin's setup.

AustinShalit, 2016-11-10:
@nightpool

nightpool, 2016-11-10:
happens to me too.

1. installed GRIP from brew cask
1. Opened GRIP
2. added a new webcam
3. enabled the "preview", watched it for a bit
4. clicked on the "remove" button
5. GRIP deadlocked.

expected result: GRIP not to deadlock ;)

SamCarlberg, 2016-11-11:
Just tested on a Windows 10 VM, can't reproduce. So it's definitely an OS X issue.

Here's what the webcam and JavaFX threads look like on OS X when removing the webcam:

```java
"Webcam 0 Service" #31 prio=5 os_prio=31 tid=0x00007fabb4abc800 nid=0x12257 runnable [0x0000700005577000]
   java.lang.Thread.State: RUNNABLE
	at org.bytedeco.javacpp.opencv_videoio.cvReleaseCapture(Native Method)
	at org.bytedeco.javacv.OpenCVFrameGrabber.stop(OpenCVFrameGrabber.java:221)
	at edu.wpi.grip.core.sources.GrabberService.shutDown(GrabberService.java:107)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:76)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- None

"JavaFX Application Thread" #13 prio=5 os_prio=31 tid=0x00007fabaf0df000 nid=0x307 waiting on condition [0x00007fff5b8b2000]
   java.lang.Thread.State: WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x000000076c24a1d0> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitUninterruptibly(AbstractQueuedSynchronizer.java:1976)
	at com.google.common.util.concurrent.Monitor.awaitUninterruptibly(Monitor.java:1102)
	at com.google.common.util.concurrent.Monitor.enterWhenUninterruptibly(Monitor.java:473)
	at com.google.common.util.concurrent.AbstractService.awaitTerminated(AbstractService.java:278)
	at com.google.common.util.concurrent.AbstractExecutionThreadService.awaitTerminated(AbstractExecutionThreadService.java:217)
	at edu.wpi.grip.core.util.service.AutoRestartingService.stopAndAwait(AutoRestartingService.java:161)
	- locked <0x000000076c23acf8> (a edu.wpi.grip.core.util.service.AutoRestartingService)
	at edu.wpi.grip.core.sources.CameraSource.stopAndAwait(CameraSource.java:289)
	at edu.wpi.grip.core.sources.CameraSource.onSourceRemovedEvent(CameraSource.java:353)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:95)
	at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:154)
	- locked <0x000000076c24c958> (a com.google.common.eventbus.Subscriber$SynchronizedSubscriber)
	at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:80)
	at com.google.common.util.concurrent.MoreExecutors$DirectExecutor.execute(MoreExecutors.java:456)
	at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:76)
	at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:119)
	at com.google.common.eventbus.EventBus.post(EventBus.java:215)
	at edu.wpi.grip.ui.pipeline.source.SourceController.delete(SourceController.java:100)
    // Nothing relevant further up the call stack
```

So the UI thread is waiting for the camera to get released, and it never gets released because `cvReleaseCapture` isn't returning.

JLLeitschuh, 2016-11-14:
Can you try `FFMpegFrameGrabber`? Or someone like @AustinShalit

SamCarlberg, 2016-11-14:
FFmpegFrameGrabber only works for video files, you can't use it for webcams. @PeterJohnson, do you think CameraServer could be an alternative?

PeterJohnson, 2016-11-14:
Yes, CameraServer will be a good alternative, but I haven't implemented IP camera support yet (ECD next week).

SamCarlberg, 2016-11-14:
I think we'd only need it for webcams, we have a custom frame grabber for IP cameras

ZeusChops, 2016-11-15:
Is this issue related to #696?

SamCarlberg, 2016-11-15:
No, #686.

PR #713 (2016-11-29): Fix removing webcams in OSX hanging
Use `stopAsync()` and a thread sleep on OS X instead of `stopAndAwait()` due to a bug in the OpenCV camera capture class.

This closes #712
closed 2016-11-29 by SamCarlberg