WPIRoboticsProjects/GRIP #717: Can't deploy deploy to roboRIO

Ubspy, 2016-11-23:
Hello, I'm trying to deploy my GRIP code to the roboRIO, but I always get the same error.
I've tried deploying it from Ubuntu 14.0.4 LTS and Windows 10.

Here's the error I get:

/usr/local/frc/JRE//bin/java -Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'
Nov 23, 2016 2:55:36 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Nov 23, 2016 2:55:36 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.5.0-rc3
Defaulting Uptime to NOIMPL due to (java.lang.UnsupportedOperationException) Implementation not available in this environment
2016-11-23 02:55:45.117:INFO::main: Logging initialized @-1ms
Nov 23, 2016 2:55:51 AM edu.wpi.grip.core.GripCoreModule onThreadException
SEVERE: Uncaught Exception on thread Thread[main,5,main]
com.google.inject.CreationException: Unable to create injector, see the following errors:

1) Error injecting method, java.lang.NoClassDefFoundError: java/beans/SimpleBeanInfo
  at edu.wpi.grip.core.serialization.Project.initialize(Unknown Source)
  at edu.wpi.grip.core.serialization.Project.class(Unknown Source)
  while locating edu.wpi.grip.core.serialization.Project
    for the 2nd parameter of edu.wpi.grip.core.http.HttpPipelineSwitcher.<init>(Unknown Source)
  at edu.wpi.grip.core.operations.network.GripNetworkModule.configure(GripNetworkModule.java:25) (via modules: com.google.inject.util.Modules$OverrideModule -> edu.wpi.grip.core.operations.network.GripNetworkModule)
  while locating edu.wpi.grip.core.http.HttpPipelineSwitcher

1 error
	at com.google.inject.internal.Errors.throwCreationExceptionIfErrorsExist(Errors.java:470)
	at com.google.inject.internal.InternalInjectorCreator.injectDynamically(InternalInjectorCreator.java:184)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:110)
	at com.google.inject.Guice.createInjector(Guice.java:99)
	at com.google.inject.Guice.createInjector(Guice.java:73)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at edu.wpi.grip.core.Main.main(Main.java:50)
Caused by: java.lang.NoClassDefFoundError: java/beans/SimpleBeanInfo
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:455)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:367)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:360)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at com.google.common.reflect.ClassPath$ClassInfo.load(ClassPath.java:274)
	at edu.wpi.grip.core.serialization.Project$$Lambda$12/20503707.apply(Unknown Source)
	at java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:193)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
	at java.util.Iterator.forEachRemaining(Iterator.java:116)
	at java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:512)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:502)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
	at edu.wpi.grip.core.serialization.Project.initialize(Project.java:64)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:483)
	at com.google.inject.internal.SingleMethodInjector$1.invoke(SingleMethodInjector.java:70)
	at com.google.inject.internal.SingleMethodInjector.inject(SingleMethodInjector.java:89)
	at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:120)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:90)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:268)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
	at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:194)
	at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
	at com.google.inject.internal.SingleParameterInjector.inject(SingleParameterInjector.java:38)
	at com.google.inject.internal.SingleParameterInjector.getAll(SingleParameterInjector.java:62)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:110)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:90)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:268)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
	at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:194)
	at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
	at com.google.inject.internal.InternalInjectorCreator$1.call(InternalInjectorCreator.java:205)
	at com.google.inject.internal.InternalInjectorCreator$1.call(InternalInjectorCreator.java:199)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1085)
	at com.google.inject.internal.InternalInjectorCreator.loadEagerSingletons(InternalInjectorCreator.java:199)
	at com.google.inject.internal.InternalInjectorCreator.injectDynamically(InternalInjectorCreator.java:180)
	... 5 more
Caused by: java.lang.ClassNotFoundException: java.beans.SimpleBeanInfo
	at java.net.URLClassLoader$1.run(URLClassLoader.java:372)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:360)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 60 more

Any ideas?

JLLeitschuh, 2016-11-28:
`java/beans/SimpleBeanInfo` may not be on the embedded JRE?

SamCarlberg, 2016-11-28:
https://github.com/WPIRoboticsProjects/GRIP/blob/master/core/src/main/java/edu/wpi/grip/core/serialization/Project.java#L59-L62

Ubspy, 2016-11-28:
@SamCarlberg How would I go about avoiding loading those classes while deploying?

SamCarlberg, 2016-11-28:
That was an oversight from a commit a while ago. You can check out #720 and build from source and it should work for you. A fix should be merged soon.

commit 26e7d961 (2016-12-02): Split project settings and app settings into separate classes (#724)

* Split project settings and app settings into separate classes

Move port validation to GripServer and common CLI to the helper class
Make project settings reflect command-line port option
No longer serializes the server port. Use the command-line option or the settings dialog to set it. Fixes #708
Enable projects to be uploaded via HTTP in UI mode

* Move *BeanInfo classes to UI module. Fixes #717

* Ignore all unknown tags in save files
closed 2016-12-02 by SamCarlberg