WPIRoboticsProjects/GRIP #718: Getting SIGBRT signal when trying to run grip

suchaHassle, 2016-11-24:


suchaHassle, 2016-11-24:
Fixed the issue, reinstalled Raspbian and redid set up.

closed 2016-11-24 by suchaHassle

JLLeitschuh, 2016-11-28:
@EdWard680 just want you to be aware of this issue in case it comes up again.

kjm16216, 2016-11-28:
Yes!  That's my error.  I just did a whole fresh install and set up yesterday and got the same result again.

suchaHassle, 2016-11-28:
@kjm16216 I recommend learning openCV and Pynetworktables for RPi and avoid using GRIP

bradamiller, 2016-11-28:
I'd recommend that you consider generating code from GRIP in Java, C++, or Python and just include the class into your program. Then you won't have to run the full GRIP executable on the target processor and you still have the advantage of the interactive features to develop your algorithm.

kjm16216, 2016-11-28:
If the process laid out by the wiki doesn't get a user there, should the wiki be updated so a future user doesn't beat his head against the wall needlessly?

Ubspy, 2016-12-13:
Guess I'm back to trying to fix errors that I get when trying to run it on the roboRIO.

SamCarlberg, 2016-12-13:
You shouldn't be trying to run GRIP on the roborio at all.

Ubspy, 2016-12-13:
@SamCarlberg If I'm not supposed to run GRIP on the roboRIO then how am I supposed to use GRIP for auto targeting?

bradamiller, 2016-12-13:
You can get the GRIP pipeline working on your laptop, then generate code to put inside your robot program. That sample program on screen steps demonstrates how you'll be able to do that as soon as everything releases. But basically it will generate a class called "PipeLine" that will have methods for running running a single pass with parameters where you can supply an image. Then there will be other generated methods that will let you get the results.

There are pull requests outstanding right now for both WPILib and GRIP that will let it generate code that actually creates a separate thread to run the vision algorithm and then call your code as a callback to retrieve the results after each processed image.

Alternatively you can run GRIP on your driver station laptop streaming the camera output to it. GRIP can process the image on the laptop, and write the resultant values to network tables which the robot program then retrieves.

You can keep trying to get GRIP running on the roboRIO or a Raspberry PI, but we're not recommending it because it was so problematic  for teams due to the varied configurations and available memory and CPU resources.

Ubspy, 2016-12-13:
Is this something new? Or have I been spending way too long trying to do the impossible?

kjm16216, 2016-12-13:
@Ubspy I spent about 2 weeks beating my head against the wall.
@bradamiller I made an update to the wiki.  I don't know if anyone cares about who updates that, but it might keep someone from beating their head against a wall like me and @Ubspy

bradamiller, 2016-12-13:
Thanks for the post. Sorry about the issue, I wish we could have gotten it working more reliably and on the wide variety of platforms that people have been trying to use.

I will say that one platform where it has worked pretty well was the Kangaroo which last year cost about $90-100 and ran windows 10. There were many teams that used that as a co-processor on their robot. I think the success there was that it was just a standard windows 10 computer with plenty of resources which was the design center for GRIP in the first place. If you're interested in trying that, I can give you an article that a mentor from FRC 1735 wrote that had all the details of making it work.

Ubspy, 2016-12-13:
Does anyone know if there's some sort of update I missed? I can't create a VisionThread object in eclipse

SamCarlberg, 2016-12-13:
It hasn't been merged yet. PRs are https://github.com/wpilibsuite/allwpilib/pull/388 and https://github.com/wpilibsuite/allwpilib/pull/399

Ubspy, 2016-12-13:
Any idea when it will be merged? And also what to update when it does?

Ubspy, 2016-12-15:
Sorry for all the stupid questions, I'm not sure how to download the prereleases, could you help me out?

Or @SamCarlberg if you know when the update will be merged that helps too.

SamCarlberg, 2016-12-15:
We're aiming for a release over the weekend

Ubspy, 2016-12-15:
@SamCarlberg Thanks!