WPIRoboticsProjects/GRIP #719: Error Invoking Method & Failed to launch JVM

FRCAluminati3555, 2016-11-25:
Downloaded Grip for windows 7 64 bit, and it will not launch; it crashes on startup. Any advise to fix this issue? It puts those errors on the screen, the ones in the title. Thanks.

JLLeitschuh, 2016-11-28:
Can you supply the error message?

Also, if you could upload your GRIP.log file you can find in your `$HOME` directory that would be really useful.

nitzanbueno, 2016-11-28:
Our team has also gotten the same issue - we installed versions 1.5.0rc1, 1.5.0rc2 and 1.5.0rc3 and all showed the same result. Also happens with 1.5.0rc3 on Windows 10 64-bit.

I ran the .jar file from the command line and it printed this:
`≡σß 28, 2016 7:56:37 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
≡σß 28, 2016 7:56:37 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.5.0-rc3
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
platform: /Windows/amd64/
Exception in Application init method
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(Unknown Source)
        at com.sun.javafx.application.LauncherImpl.launchApplication(Unknown Source)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)
Caused by: java.lang.RuntimeException: Exception in Application init method
        at com.sun.javafx.application.LauncherImpl.launchApplication1(Unknown Source)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$155(Unknown Source)
        at java.lang.Thread.run(Unknown Source)
Caused by: edu.wpi.grip.core.exception.GripServerException: Could not start Jetty server
        at edu.wpi.grip.core.http.GripServer.start(GripServer.java:158)
        at edu.wpi.grip.ui.Main.init(Main.java:100)
        ... 3 more
Caused by: java.net.BindException: Address already in use: bind
        at sun.nio.ch.Net.bind0(Native Method)
        at sun.nio.ch.Net.bind(Unknown Source)
        at sun.nio.ch.Net.bind(Unknown Source)
        at sun.nio.ch.ServerSocketChannelImpl.bind(Unknown Source)
        at sun.nio.ch.ServerSocketAdaptor.bind(Unknown Source)
        at org.eclipse.jetty.server.ServerConnector.open(ServerConnector.java:326)
        at org.eclipse.jetty.server.AbstractNetworkConnector.doStart(AbstractNetworkConnector.java:80)
        at org.eclipse.jetty.server.ServerConnector.doStart(ServerConnector.java:244)
        at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68)
        at org.eclipse.jetty.server.Server.doStart(Server.java:384)
        at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68)
        at edu.wpi.grip.core.http.GripServer.start(GripServer.java:156)
        ... 4 more
Exception running application edu.wpi.grip.ui.Main`

Here is the log file, although it doesn't seem too helpful:
`נוב 28, 2016 7:56:37 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
נוב 28, 2016 7:56:37 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.5.0-rc3
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Module execution: 290ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: TypeListeners & ProvisionListener creation: 2ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Scopes creation: 3ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Converters creation: 0ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Binding creation: 18ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Module annotated method scanners creation: 1ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Private environment creation: 1ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Injector construction: 0ms
נוב 28, 2016 7:56:37 PM com.google.inject.internal.util.Stopwatch resetAndLog
[... 1146 lines cut ...]
FINE: Injector construction: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Binding initialization: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Binding indexing: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Collecting injection requests: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Binding validation: 1ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Static validation: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Instance member validation: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Provider verification: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Static member injection: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Instance injection: 1ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Preloading singletons: 0ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Instance injection: 276ms
נוב 28, 2016 7:56:38 PM com.google.inject.internal.util.Stopwatch resetAndLog
FINE: Preloading singletons: 531ms`

SamCarlberg, 2016-11-28:
You either have another instance of GRIP running without changing the HTTP port in the settings, or there's another app using that port.

nitzanbueno, 2016-11-28:
I ran GRIP first thing after a restart and on two different computers and the same problem arose, so I doubt that's the reason. Maybe the code accidentally starts the server twice?

SamCarlberg, 2016-11-28:
Then an app starts on login that uses port 8080. You should be able to go to `http://localhost:8080` in a browser and see a webpage.

FRCAluminati3555, 2016-11-28:
Here's the log file:

Nov 28, 2016 3:08:23 PM java.util.logging.LogManager$RootLogger log

CONFIG: Configuration done.

Nov 28, 2016 3:08:23 PM java.util.logging.LogManager$RootLogger log

CONFIG: GRIP Version: 1.5.0-rc3

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Module execution: 1455ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: TypeListeners & ProvisionListener creation: 9ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Scopes creation: 5ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Converters creation: 2ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Binding creation: 107ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Module annotated method scanners creation: 2ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Private environment creation: 3ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Injector construction: 1ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Binding initialization: 170ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Binding indexing: 2ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Collecting injection requests: 1ms

Nov 28, 2016 3:08:25 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Binding validation: 37ms

[... 2761 lines cut ...]
resetAndLog

FINE: Binding initialization: 1ms

Nov 28, 2016 3:08:26 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Binding indexing: 1ms

Nov 28, 2016 3:08:26 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Collecting injection requests: 1ms

Nov 28, 2016 3:08:26 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Binding validation: 0ms

Nov 28, 2016 3:08:26 PM com.google.inject.internal.util.Stopwatch
resetAndLog

FINE: Static validation: 2ms

Nov 28, 2

SamCarlberg, 2016-11-28:
I'm going to delete the log text, there's nothing useful and it makes the page very difficult to read.

GRIP defaults to port 8080, whatever's running on 49970 isn't us.

FRCAluminati3555, 2016-11-28:
Go for it, and going to http://localhost:8080 results in a 404 error message

SamCarlberg, 2016-11-28:
What's the message/can you post a screenshot? I want to see if it's GRIP's 404 or a different one

FRCAluminati3555, 2016-11-28:
[image]

FRCAluminati3555, 2016-11-28:
[image]

FRCAluminati3555, 2016-11-28:
I wasn't sure if this was what you were looking for, but running these will just bring up the loading bar, they don't bring up the error panels, but nothing else happens, nothing else is brought up.

SamCarlberg, 2016-11-28:
You need to run the jar with `java -jar C:\Users\frcAluminati\...\GRIP-1.5.0-rc3-x64.jar`. But another app (I don't know what) is running on port 8080 and is blocking GRIP from using it.

The master branch has support for setting the HTTP port from the command line. You can download or clone it, then run `gradlew :ui:run -Pargs="--port <port number>"` in the root GRIP folder.

You can also change the server port in the save file, there should be an entry for it near the bottom of the file

FRCAluminati3555, 2016-11-28:
Which file do i go to to change the server port, and what should i change it to?

[image]

SamCarlberg, 2016-11-28:
You need to set it on the command line

1. Run powershell (Windows key -> type "powershell")
2. Run `cd` and give it the path to the GRIP folder you downloaded (e.g. `cd C:\Users\frcAluminati\Downloads\GRIP-master`)
3. Run `gradlew :ui:run -Pargs="--port 3555"` (or whatever port number you want)

FRCAluminati3555, 2016-11-28:
Okay I did that, but then how do I run it?

SamCarlberg, 2016-11-28:
The `gradlew` command runs it.

FRCAluminati3555, 2016-11-28:
It says that gradlew isn't a recognized name of cmdlet, I don't have much experience with running things through the command line.

SamCarlberg, 2016-11-28:
Sorry, I meant to use the old command line. Run "cmd" instead of "poewrshell" in step 1 and that should work for you.

FRCAluminati3555, 2016-11-28:
Thanks, it worked and is now running, do we have to run that command every time to use this?

closed 2016-11-28 by FRCAluminati3555

SamCarlberg, 2016-11-28:
For now, yes. You can write a batch script that will run GRIP with command line arguments so you don't have to run it from the command line every time.

We might add a configuration file at some point, but that wouldn't happen for a while.

For the batch script:

```batch
C:\Users\frcAluminati\AppData\Local\GRIP\GRIP.exe --port 3555
```

Should run GRIP on port 3555. Just create a `GRIP.bat` file and paste that line into it and you should be good to go.

FRCAluminati3555, 2016-11-28:
Thanks for the help!