WPIRoboticsProjects/GRIP #721: Don't crash GRIP if there is already a process bound to port 8080

JLLeitschuh, 2016-11-28:
Grip should instead startup with a warning and the internal server disabled, or perhaps with a dialog that lets you pick a different port to start the server on.

SamCarlberg, 2016-11-28:
Can't disable the server. Dialog might work. How would you handle this for the headless case?

JLLeitschuh, 2016-12-20:
Uhh... For the headful case just shut down the server and show a warning?

AustinShalit, 2016-12-20:
:+1: I agree that this should be fixed.

Wazzaps, 2017-01-03:
I would recommend to assign a random unused port and use that, and show a warning that the port is used. Or have the port configurable

SamCarlberg, 2017-01-03:
I forgot to close this one. The port's been changed to 2084 from 8080, and allows the user to continue without the HTTP functionality if the port is already in use.

closed 2017-01-03 by SamCarlberg