WPIRoboticsProjects/GRIP #735: tests fail due to some issues.

fly6464, 2016-12-02:
Hi, 

I am working on windows 7 and tried to run ui tests, but all of the tests under the \tool folder fail. There are two errors 1) java.lang.UnsatisfiedLinkError: no opencv_java310 in java.library.path and 2) java.lang.NoClassDefFoundError: Could not initialize class edu.wpi.grip.ui.codegeneration.AbstractGenerationTesting. I guess there is something wrong with opencv. Could anyone give some hints? Do I need to install opencv on my local machine?

SamCarlberg, 2016-12-02:
What command are you using to run the tests?

fly6464, 2016-12-02:
Thanks for replying. I just ran the tests using IntelliJ IDEA. All the tests under the \tool folder fail, but all other tests can pass including GRIP/core/src/test/java/edu/wpi/grip/core/operations/opencv/AddOperationTest.java. Do you think all the needed opencv dlls are automatically downloaded and included in the project or I need to prepare by myself? Actually I cannot find opencv_java310 on my local machine.

SamCarlberg, 2016-12-02:
This should only happen if you're running code generation tests with `-Pgeneration` or `-Pgenonly`. Code generation tests compile and run generated code to make sure it's functional. You will need OpenCV installed to do this, as well as the OpenCV python bindings.

If you're not making changes to code generation, I strongly recommend not running those tests to avoid the dependency headaches.

We have an applicatinon [here](https://github.com/WPIRoboticsProjects/opencv-installer) that we use to install barebones OpenCV JNI and C++ bindings. Since you're on Windows, they should show up in `C:\Users\<user>\OpenCV`. You'll need to have this directory on the PATH environment variable. Python bindings can be installed with pip (`pip install opencv`). This _should_ be enough to get it working.

fly6464, 2016-12-02:
Thanks a lot. I didn't specify -Pgeneration nor -Pgenonly when ran tests. Actually, I am a new user to gradle. Can you give me a step-by-step installation guidance of the OpenCV project? Appreciate it!