WPIRoboticsProjects/GRIP #736: ntcore dependency required by grip-1.5.0-x64.deb

smile-on, 2016-12-04:
after installing grip-1.5.0-x64.deb on Ubuntu execution of GRIP shows splash screen and fails with error
```
platform: /Linux/amd64/
Exception in Application init method
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
Caused by: java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at edu.wpi.first.wpilibj.networktables.NetworkTablesJNI.<clinit>(NetworkTablesJNI.java:57)
	at edu.wpi.grip.core.operations.network.networktables.NTManager.<init>(NTManager.java:78)
```
Could not find Grip's prerequisites on wiki. Is ntcore the only pre-requirement? Not sure what additional packages need to be installed to run GRIP.
[grip-error.log.zip](https://github.com/WPIRoboticsProjects/GRIP/files/629252/grip-error.log.zip)

SamCarlberg, 2016-12-04:
You don't need to have anything installed to run GRIP. I just created a new Ubuntu 14.04 VM and downloaded and ran GRIP with no issues.

smile-on, 2016-12-05:
Something is different in my installation on Ubuntu 12.05.1.  ntcore lib file is clearly missing. Could you check where **ntcore** file comes from in your installation?

SamCarlberg, 2016-12-05:
The .so should be in

`/opt/grip/app/lib/core-1.5.0-all.jar/Linux/x86/libntcore.so`

smile-on, 2016-12-22:
after Linux/amd64/libntcore.so was manually extracted and put under java.library.path the error message got new error on gcc runtime
```
Caused by: com.google.inject.ProvisionException: Unable to provision, see the following errors:
1) Error injecting constructor, java.lang.UnsatisfiedLinkError: /opt/GRIP/runtime/lib/amd64/libntcore.so: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.19' not found (required by /opt/GRIP/runtime/lib/amd64/libntcore.so)
  at edu.wpi.grip.core.operations.network.networktables.NTManager.<init>(Unknown Source)
```
I do have /x86_64-linux-gnu/libstdc++.so.6 library from **libstdc++6** package version **4.6.3-1ubuntu5** (GNU Standard C++ Library v3). It looks like *libntcore.so has dependency on some specific glibc*.

smile-on, 2016-12-22:
GRIP has ntcore and opencv that do require **gcc version 4.8** or higher. Ubuntu 12 is just too old (gcc version 4.6.3). Running update on gcc from [PPA ubuntu toolchain R](https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test) completely fixed this problem.
```
$ sudo add-apt-repository ppa:ubuntu-toolchain-r/test
$ sudo apt-get update
$ sudo apt-get upgrade
```
Thank you.

closed 2016-12-22 by smile-on