WPIRoboticsProjects/GRIP #737: GRIP runaway CPU when webcam isn't available

virtuald, 2016-12-04:
Lots of this happens over and over again when I add a webcam that isn't plugged in:

```
SEVERE: [CameraSource] Failed from: STARTING
edu.wpi.grip.core.sources.GrabberService$GrabberServiceException: [Webcam 0] Failed to start
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:54)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:54)
	at com.google.common.util.concurrent.Callables$3.run(Callables.java:100)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: cvCreateCameraCapture() Error: Could not create camera capture.
	at org.bytedeco.javacv.OpenCVFrameGrabber.start(OpenCVFrameGrabber.java:179)
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:52)
	... 3 more
```

It seems like the pipeline is running, even though I didn't ask it to run? (shows a stop icon on the bottom)

SamCarlberg, 2016-12-04:
If it can't connect to the webcam given, it'll immediately try to reconnect. It's independent of the pipeline running.

virtuald, 2016-12-04:
Then it shouldn't log the error?

It seems like there should be some delay before trying again, otherwise it just runs the CPU at a million percent, which is annoying for a variety of reasons.

SamCarlberg, 2016-12-04:
@JLLeitschuh do you know why there's no delay between retries?

SamCarlberg, 2016-12-05:
I talked to Jonathan, he says it's to make it more forgiving for FRC teams so they can plug in webcams on the robot after they start GRIP. Might not be so useful now that we're moving away from the deploy functionality.

virtuald, 2016-12-05:
Retrying is fine. Retrying without a delay is not fine.

bradamiller, 2016-12-05:
I have to agree. Flooding the system with retries seems way to severe. No more than once a second and maybe less seems just fine. Having grip start going a second after the camera is plugged in seems plenty responsive.

PR #759 (2016-12-29): Limit camara connection retries to 50Hz
Fixes #737
closed 2016-12-29 by SamCarlberg