WPIRoboticsProjects/GRIP #739: Update Publish Video format

AustinShalit, 2016-12-06:
We need to update the `Publish Video` step to use the new stream format.

As reported by @PeterJohnson

JLLeitschuh, 2016-12-08:
This will be a breaking change but I don't think this requires a major version bump Just document it well.

bradamiller, 2016-12-08:
If making it an incompatible change is a concern, add it as a different video publisher and keep the old stuff.

AustinShalit, 2017-02-15:
It looks like this is higher priority because what we have now is broken.

JLLeitschuh, 2017-02-27:
Do we have any need to support both the old and new format or do we only need to support the new format moving forward?

bradamiller, 2017-02-27:
What's the old format? Is that the NI stream format? But right now it's broken, I agree with Austin that it should be fixed, even with a separate publisher then deprecating the old one.