WPIRoboticsProjects/GRIP #740: GRIP not running correctly on RIO

Ubspy, 2016-12-06:
Hello, everyone! I seem to be having a problem with the NetworkTables on the roborRIO.

I have successfully deployed my code to the roboRIO, but it doesn't seem to want to run correctly for some reason. If I run the code on the roboRIO, nothing is published to the NetworkTables. Even though in the console I get: "INFO: SUCCESS! The project is running in headless mode."

If I run the GRIP code on my laptop however, it publishes fine to the NetworkTables. Do you guys know what might be the source of the problem?

SamCarlberg, 2016-12-06:
You need to launch GRIP with the save file to run

```bash
$ /home/lvuser/bin/java -jar /path/to/grip.jar --file /path/to/save.grip
```

---

We're leaning away from the deploy function altogether in favor of generating code you can run in your robot program. It's faster, uses less memory, and is less of a headache to get working. Have you tried that instead?

JLLeitschuh, 2016-12-06:
@Ubspy Where are you running your network tables server?

Ubspy, 2016-12-07:
@SamCarlberg That just threw me a huge error.

@JLLeitschuh I should be running the network tables server on the roboRIO, when I use GRIP on my laptop it publishes the values correctly. But not on the RIO.

Ubspy, 2016-12-08:
Also @SamCarlberg I was running "/usr/local/frc/JRE/bin/java -jar '/home/lvuser/grip.jar' '/home/lvuser/project.grip'" if that clarifies anything for you.

SamCarlberg, 2016-12-08:
That command won't work. You need to use the `--file` option to specify the project file

```bash
$ /usr/local/frc/JRE/bin/java -jar /home/lvuser/grip.jar --file /home/lvuser/project.grip
```

Ubspy, 2016-12-08:
@SamCarlberg that gives me a massive error

SamCarlberg, 2016-12-08:
Can you post it? Preferably as a [gist](http://gist.github.com) if it's long

Ubspy, 2016-12-08:
@SamCarlberg here's the error I got: https://gist.github.com/Ubspy/e395c7aed95ba858fc986296cc800f0d

SamCarlberg, 2016-12-08:
Your pipeline was started, but OpenCV couldn't understand the video format ([these lines](https://gist.github.com/Ubspy/e395c7aed95ba858fc986296cc800f0d#file-grip-error-L117-L124)).

What camera are you using, and how is it connected to the roborio?

Ubspy, 2016-12-08:
@SamCarlberg we're using the Microsoft LifeCam HD-3000, and it's connected via USB.

SamCarlberg, 2016-12-08:
Any other USB devices plugged in?

Ubspy, 2016-12-08:
No, just the camera.

Ubspy, 2016-12-08:
@SamCarlberg No, just the camera.

But it seems that if I plug that camera into my laptop it works just fine. It still publishes all the values to network tables.

Ubspy, 2016-12-12:
@SamCarlberg Is there some kind of driver I need for the camera? Because it works fine on my laptop, but the camera doesn't seem to want to start on the RIO.

Ubspy, 2016-12-14:
Looks like the new version of GRIP isn't supposed to run on the RIO, so that solves this issue.

closed 2016-12-14 by Ubspy