WPIRoboticsProjects/GRIP #752: Can't run GRIP on rPi2

Ubspy, 2016-12-13:
When I try to run grip with "env LD_LIBRARY_PATH=/home/pi/vision/grip:$LD_LIBRARY_PATH java -jar /home/pi/vision/grip/grip.jar /home/pi/vision/grip/project.grip" I get this error:

"terminate called after throwing an instance of 'std::logic_error'
    what(): basic_string::_S_contruct null not valid
Aborted"

Any ideas?

kjm16216, 2016-12-13:
https://github.com/WPIRoboticsProjects/GRIP/issues/718

SamCarlberg, 2016-12-13:
@kjm16216 Thanks for pointing that out

closed 2016-12-13 by SamCarlberg

kjm16216, 2016-12-13:
@Ubspy I'm working on a wrapper so we can push a GRIP generated filter onto a pi.  Want in?  How are you with network tables?

Ubspy, 2016-12-13:
@kjm16216 I'm only in high school, I'm not sure I'd be too much help.

As for network tables, I understand the concept, just not entirely how it works.

kjm16216, 2016-12-13:
High schoolers run circles around me on here every day.
I've got a Pi3 that will take an image from the Pi camera module and run it through a GRIP generated pipeline, then save it to a file.
Next step is to replace the save with one of the data reports (blobs, contours, etc) and output that to a network table.

Ubspy, 2016-12-13:
@kjm16216 If you'd like some help, sure.

Although I don't feel that competent, that won't stop me from trying. Also a small warning: with finals week coming up my schedule is somewhat booked.

bradamiller, 2016-12-13:
Sorry about your dissapointment about making GRIP run well on the PI. We're also agree with you, but have been having too much trouble getting GRIP working reliably in those environments. So code generation seemed like a good solution, much less resource intensive, and easier to merge into your existing robot program.

We're trying to put together some samples and a pre-built vision of OpenCV for the PI3. The thought is to use the python generated code since it is pretty concise and there is a python version of NetworkTables so that is one less thing to build and have to link.

kjm16216, 2016-12-13:
I'm in Java.