WPIRoboticsProjects/GRIP #765: Pi 3 deploy issues

Cad321, 2016-12-31:
While trying to delpoy to my Pi 3, I run into the following error(s?):

`/usr/bin/java -Xmx750m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -jar '/home/pi/vision/grip/grip.jar' '/home/pi/vision/grip/project.grip'
Dec 31, 2016 3:31:14 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Dec 31, 2016 3:31:14 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 0.0.0
2016-12-31 03:31:16.990:INFO::main: Logging initialized @2778ms
platform: /Linux/arm/
Dec 31, 2016 3:31:18 AM edu.wpi.grip.core.GripCoreModule onThreadException
SEVERE: Uncaught Exception on thread Thread[main,5,main]
com.google.inject.ProvisionException: Unable to provision, see the following errors:

1) Error injecting constructor, java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
  at edu.wpi.grip.core.operations.network.networktables.NTManager.<init>(Unknown Source)
  at edu.wpi.grip.core.operations.network.networktables.NTManager.class(Unknown Source)
  while locating edu.wpi.grip.core.operations.network.networktables.NTManager
  while locating edu.wpi.grip.core.operations.network.MapNetworkPublisherFactory annotated with @com.google.inject.name.Named(value=ntManager)
    for the 2nd parameter of edu.wpi.grip.core.operations.Operations.<init>(Unknown Source)
  at edu.wpi.grip.core.operations.Operations.class(Unknown Source)
  while locating edu.wpi.grip.core.operations.Operations
    for field at edu.wpi.grip.core.Main.operations(Unknown Source)
  while locating edu.wpi.grip.core.Main

1 error
	at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1028)
	at com.google.inject.internal.InjectorImpl.getInstance(InjectorImpl.java:1054)
	at edu.wpi.grip.core.Main.main(Main.java:58)
Caused by: java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1864)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at edu.wpi.first.wpilibj.networktables.NetworkTablesJNI.<clinit>(NetworkTablesJNI.java:59)
	at edu.wpi.grip.core.operations.network.networktables.NTManager.<init>(NTManager.java:78)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
	at com.google.inject.internal.DefaultConstructionProxyFactory$ReflectiveProxy.newInstance(DefaultConstructionProxyFactory.java:126)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:111)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:90)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:268)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
	at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:194)
	at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
	at com.google.inject.internal.FactoryProxy.get(FactoryProxy.java:56)
	at com.google.inject.internal.SingleParameterInjector.inject(SingleParameterInjector.java:38)
	at com.google.inject.internal.SingleParameterInjector.getAll(SingleParameterInjector.java:62)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:110)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:90)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:268)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
	at com.google.inject.internal.SingletonScope$1.get(SingletonScope.java:194)
	at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:41)
	at com.google.inject.internal.SingleFieldInjector.inject(SingleFieldInjector.java:54)
	at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:132)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:120)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:90)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:268)
	at com.google.inject.internal.InjectorImpl$2$1.call(InjectorImpl.java:1019)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1085)
	at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1015)
	... 2 more`

Based on suggestions made in another isssue, I tried editing the `.bash_profile` file and inserting this: `set LD_LIBRARY_PATH=/home/pi/vision/grip:$LD_LIBRARY_PATH`.

However whenever I try to edit the file (or search for it) I can not seem to find it. Any suggestions?

blucoat, 2017-01-07:
For one, make sure that you have ntcore installed on the pi, as the one that ships with grip is built for armel, not armhf (making it incompatible).  Once you have that installed, say to /usr/local/lib, you need to add -Djava.library.path=/usr/local/lib to the jvm arguments when you deploy.  I spent some time last summer getting GRIP to deploy to the pi and am revisiting it now, so if you get stuck on anything feel free to ask.

closed 2017-01-08 by SamCarlberg

kjm16216, 2017-01-09:
@blucoat Could I impose on you for a little help getting GRIP/NT working on a pi?

SamCarlberg, 2017-01-09:
WPI has networktables builds for the pi 3 now. Most current version [here](http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilib/networktables/java/NetworkTables/3.1.5/NetworkTables-3.1.5-arm-raspbian.jar)

blucoat, 2017-01-09:
@kjm16216 What I ended up doing was updating the version of javacv which GRIP depends on to a newer one which supports armhf, and building a custom version of GRIP using that.  To install NT, I just built from source on the pi, but it looks like now there's a better way to do it.  Also, I ran into some strange issue with javacv [not detecting the architecture right](https://github.com/bytedeco/javacpp/issues/105).  It looks like it's been fixed, but I haven't had a chance to test it yet.

Before you put too much effort into that, it looks like the GRIP developers really want you to use autogenerated code, rather than deploying now.  Our team bought a pi and I really would like to isolate vision code from the robot code, both for performance and safety reasons, but as of right now I'm most concerned with getting something that works quickly.  It looks like this year's version of WPILib (at least the Java version, I haven't looked at the others) includes opencv and a big improvement over the old CameraServer.  It looks like it's the Java bindings that come with OpenCV, not JavaCV (which is much better IMO), but it's still better than being left to install them yourself.  For right now at least, I think we're going to use this approach.

Maybe there's a good way to run autogenerated code on the pi, without having to deploy GRIP directly, or maybe with the new changes deploying to the pi will become much easier.  Either way, I'd love to see more support for offloading vision code to a pi, but until then, I don't think that's the path I'll be taking.  For the next few weeks I'll be mostly dedicated to working on my team's code, and then I'll be going back to school, but after that I might try to seriously work on a better solution for the pi.  In the meantime, I can answer any specific questions/problems you run into.

kjm16216, 2017-01-10:
@blucoat Here's where I'm at: I can open the Raspberry Pi Camera module, capture an image, perform a GRIP generated filter (from autogenerated code), and display the values of the report to the console.
What I am struggling with now is getting those values out onto the network to be picked up by the Rio.
Our mechanical team has also designed a 3D printable Pi/Cam mount which incorporates a light ring.
Think we can help each other out?
Using a Pi3 and Java.

Specifically, right now I have code that is supposed to put it out on the network, before the holidays, it would give an error both in Eclipse when I tried to run and on the pi.  Then I committed the cardinal sin of changing 2 things at once.
I added NetworkTablesSources to my build path in Eclipse, and I updated to the 2017 libraries, without testing in between.  So I have a new error:
`Exception in thread "main" java.lang.UnsatisfiedLinkError: no wpilibJavaJNI in java.library.path
	at java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.lang.Runtime.loadLibrary0(Unknown Source)
	at java.lang.System.loadLibrary(Unknown Source)
	at edu.wpi.first.wpilibj.hal.JNIWrapper.<clinit>(JNIWrapper.java:22)
	at edu.wpi.first.wpilibj.RobotBase.initializeHardwareConfiguration(RobotBase.java:171)
	at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:194)`
That's in an Eclipse run.  I haven't made the jar and run on the pi since adding NTSource.

blucoat, 2017-01-10:
@kjm16216 To avoid spamming this issue, want to switch to email?

Just to make sure, you're talking about a deploy to the RIO from eclipse?  It seems that the code can't find the native libraries for wpilib (not even networktables).  If you make a new blank project and try to deploy it, does it work?

kjm16216, 2017-01-10:
Sure, sent!  Thanks.