WPIRoboticsProjects/GRIP #767: Unable to run GRIP 1.5.1-rc1

sciencewhiz, 2017-01-02:
After installing GRIP-1.5.1-rc1-x64.exe, GRIP fails to start. The popup that is displayed is shown below.

[image]

SamCarlberg, 2017-01-02:
This seems to be a consistent problem with the JavaFX app installer. Did you run the uninstaller for the previous version before installing 1.5.1-rc1?

sciencewhiz, 2017-01-02:
No

SamCarlberg, 2017-01-02:
You should do that. You can uninstall GRIP with the "Programs and Features" Windows app.

I'll look into seeing if it's possible to get the installer to remove the previous version

sciencewhiz, 2017-01-02:
I unistalled GRIP, deleted the GRIP folder (there were still a few items in it), then reinstalled GRIP rc1 and had the same issue

bradamiller, 2017-01-02:
The Mac version is working for me.

bradamiller, 2017-01-02:
running the same version in parallels (windows 10 on my Mac), GRIP started, but the webcam crashed - it doesn't seem to be shared in parallels.