WPIRoboticsProjects/GRIP #768: Added webcam -> cvResize -> cvtColor(BGR2BGRA)

bradamiller, 2017-01-02:
Created the pipeline webcam -> cvResize -> cvtColor(BGR2BGRA), and it crashed.

SamCarlberg, 2017-01-02:
Error message?

nightpool, 2017-01-03:
I can confirm this on mac. It took the entire build down, I didn't get the "this exception was thrown" screen, so that needs to be fixed first. Here's the error, from the console:

```
java.lang.AssertionError: Only 3-channel BGR images or single-channel grayscale images can be converted
        at edu.wpi.grip.ui.util.ImageConverter.convert(ImageConverter.java:55)
        at edu.wpi.grip.ui.preview.ImageSocketPreviewView.lambda$null$0(ImageSocketPreviewView.java:31)
        at edu.wpi.grip.ui.preview.ImageSocketPreviewView$$Lambda$680/778946230.run(Unknown Source)
        at edu.wpi.grip.ui.util.GripPlatform.runAsSoonAsPossible(GripPlatform.java:51)
        at edu.wpi.grip.ui.preview.ImageSocketPreviewView.lambda$convertImage$1(ImageSocketPreviewView.java:30)
        at edu.wpi.grip.ui.preview.ImageSocketPreviewView$$Lambda$679/201664973.accept(Unknown Source)
        at java.util.Optional.ifPresent(Optional.java:159)
        at edu.wpi.grip.ui.preview.ImageSocketPreviewView.convertImage(ImageSocketPreviewView.java:29)
        at edu.wpi.grip.ui.preview.ImageBasedPreviewView.resize(ImageBasedPreviewView.java:63)
        at edu.wpi.grip.ui.preview.PreviewsController.lambda$onSocketPreviewChanged$8(PreviewsController.java:134)
        at edu.wpi.grip.ui.preview.PreviewsController$$Lambda$678/459878562.run(Unknown Source)
        at com.sun.javafx.application.PlatformImpl.runAndWait(PlatformImpl.java:317)
        at com.sun.javafx.application.PlatformImpl.runAndWait(PlatformImpl.java:307)
        at edu.wpi.grip.ui.preview.PreviewsController.onSocketPreviewChanged(PreviewsController.java:128)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:91)
        at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:150)
        at com.google.common.eventbus.Subscriber$1.run(Subscriber.java:76)
        at com.google.common.util.concurrent.MoreExecutors$DirectExecutor.execute(MoreExecutors.java:435)
        at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:71)
        at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:116)
        at com.google.common.eventbus.EventBus.post(EventBus.java:217)
        at edu.wpi.grip.core.sockets.OutputSocketImpl.setPreviewed(OutputSocketImpl.java:45)
        at edu.wpi.grip.ui.pipeline.OutputSocketController.lambda$new$0(OutputSocketController.java:61)
        at edu.wpi.grip.ui.pipeline.OutputSocketController$$Lambda$576/1363243807.invalidated(Unknown Source)
        at com.sun.javafx.binding.ExpressionHelper$SingleInvalidation.fireValueChangedEvent(ExpressionHelper.java:137)
        at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
        at javafx.beans.property.BooleanPropertyBase.fireValueChangedEvent(BooleanPropertyBase.java:103)
        at javafx.beans.property.BooleanPropertyBase.markInvalid(BooleanPropertyBase.java:110)
        at javafx.beans.property.BooleanPropertyBase.set(BooleanPropertyBase.java:144)
        at javafx.scene.control.ToggleButton.setSelected(ToggleButton.java:149)
        at javafx.scene.control.ToggleButton.fire(ToggleButton.java:255)
        at com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(ButtonBehavior.java:182)
        at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:96)
        at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:89)
        at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
[... cut ...]
        at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:116)
        at com.google.common.eventbus.EventBus.post(EventBus.java:217)
        at edu.wpi.grip.core.GripCoreModule.onThreadException(GripCoreModule.java:199)
        at edu.wpi.grip.core.GripCoreModule$$Lambda$55/1861073427.uncaughtException(Unknown Source)
        at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
        at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
        at com.sun.glass.ui.Application.reportException(Application.java:453)
```

SamCarlberg, 2017-01-03:
Windows 10: cannot reproduce. I _do_ get an exception when trying to preview the `cvtColor` output, but that's because it's a 4-channel image when only 3-channel images can be previewed. This should probably just be changed to show a warning dialog

---

Thread[JavaFX Application Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.UnsupportedOperationException: Only 1 or 3 channel images can be shown, tried to show a 4 channel image
	at edu.wpi.grip.ui.util.ImageConverter.convert(ImageConverter.java:118)
	at edu.wpi.grip.ui.preview.ImageSocketPreviewView.lambda$null$0(ImageSocketPreviewView.java:31)
	at edu.wpi.grip.ui.util.GripPlatform.lambda$onJavaFXRunnerEvent$0(GripPlatform.java:74)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | 1.5.1-rc1
java.version | 1.8.0_101
javafx.version | 8.0.101
os.name | Windows 10
os.version | 10.0
os.arch | amd64

SamCarlberg, 2017-01-03:
Interestingly, it looks like it's running the pipeline in the JavaFX thread... @bradamiller, does this happen when running Windows in parallels?

bradamiller, 2017-01-03:
I'll have to test it tomorrow when I'm at school with a web cam that I can assign to the VM. I have the camera shared with windows, but it doesn't seem to be working in GRIP. If I plug in a real USB camera and assign it, I expect it will work.

SamCarlberg, 2017-01-03:
You shouldn't have to use a webcam for this, any image file should be fine

bradamiller, 2017-01-03:
I get the exception, but it doesn't crash. Maybe at some point we can make it check for things with mismatched numbers of channels.

SamCarlberg, 2017-01-03:
Oh, I misread your stacktrace. I had the same error. This is definitely something that can be improved.

JLLeitschuh, 2017-01-03:
Note, `assert` is turned off in releases.
You will only get assertion failures from using the `assert` keyword in development.

PR #821 (2017-02-13): Don't allow non-previewable values to be previewed
Fixes #768
commit 0eb6ce44 (2017-02-13): Don't allow non-previewable values to be previewed (#821)

Fixes #768
closed 2017-02-13 by SamCarlberg