WPIRoboticsProjects/GRIP #779: Unable to run GRIP `:ui:run` on Windows 10

rlee287, 2017-01-07:
While the executable installer works perfectly, the `gradlew.bat :ui:run` command fails to build the GRIP project on Windows 10. The build fails on `v1.5.1`, but the executable from the releases page works fine.

Running `gradlew.bat --stacktrace :ui:run` produces the following exception log:

```
<---snipped out earlier tasks that completed normally --->
:ui:run
Jan 06, 2017 11:00:27 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 06, 2017 11:00:27 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: v1.5.1
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in Application init method
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
        at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:767)
Caused by: java.lang.RuntimeException: Exception in Application init method
        at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:912)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$155(LauncherImpl.java:182)
        at java.lang.Thread.run(Thread.java:745)
Caused by: com.google.inject.CreationException: Unable to create injector, see the following errors:

1) Error injecting method, java.lang.NoClassDefFoundError: edu/wpi/grip/ui/GRIPUIModule$1 (wrong name: edu/wpi/grip/ui/GripUiModule$1)
  at edu.wpi.grip.core.serialization.Project.initialize(Unknown Source)
  at edu.wpi.grip.core.serialization.Project.class(Unknown Source)
  while locating edu.wpi.grip.core.serialization.Project
    for the 2nd parameter of edu.wpi.grip.core.http.HttpPipelineSwitcher.<init>(Unknown Source)
  at edu.wpi.grip.core.operations.network.GripNetworkModule.configure(GripNetworkModule.java:25) (via modules: com.google.inject.util.Modules$OverrideModule -> edu.wpi.grip.core.operations.network.GripNetworkModule)
  while locating edu.wpi.grip.core.http.HttpPipelineSwitcher

1 error
        at com.google.inject.internal.Errors.throwCreationExceptionIfErrorsExist(Errors.java:470)
        at com.google.inject.internal.InternalInjectorCreator.injectDynamically(InternalInjectorCreator.java:184)
        at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:110)
        at com.google.inject.Guice.createInjector(Guice.java:99)
        at com.google.inject.Guice.createInjector(Guice.java:73)
        at com.google.inject.Guice.createInjector(Guice.java:62)
        at edu.wpi.grip.ui.Main.init(Main.java:91)
        at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:841)
        ... 2 more
Caused by: java.lang.NoClassDefFoundError: edu/wpi/grip/ui/GRIPUIModule$1 (wrong name: edu/wpi/grip/ui/GripUiModule$1)
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
        at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
        at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
        at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        at com.google.common.reflect.ClassPath$ClassInfo.load(ClassPath.java:306)
        at java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:193)
        at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
        at java.util.Iterator.forEachRemaining(Iterator.java:116)
        at java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
        at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
[... 65 lines cut ...]
        at org.gradle.execution.taskgraph.AbstractTaskPlanExecutor$TaskExecutorWorker.run(AbstractTaskPlanExecutor.java:55)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
        at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
Caused by: org.gradle.process.internal.ExecException: Process 'command 'C:\Program Files\Java\jdk1.8.0_102\bin\java.exe'' finished with non-zero exit value 1
        at org.gradle.process.internal.DefaultExecHandle$ExecResultImpl.assertNormalExitValue(DefaultExecHandle.java:369)
        at org.gradle.process.internal.DefaultJavaExecAction.execute(DefaultJavaExecAction.java:31)
        at org.gradle.api.tasks.JavaExec.exec(JavaExec.java:74)
        at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
        at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$StandardTaskAction.doExecute(DefaultTaskClassInfoStore.java:141)
        at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$StandardTaskAction.execute(DefaultTaskClassInfoStore.java:134)
        at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$StandardTaskAction.execute(DefaultTaskClassInfoStore.java:123)
        at org.gradle.api.internal.AbstractTask$TaskActionWrapper.execute(AbstractTask.java:624)
        at org.gradle.api.internal.AbstractTask$TaskActionWrapper.execute(AbstractTask.java:607)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeAction(ExecuteActionsTaskExecuter.java:95)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:76)
        ... 16 more

BUILD FAILED

Total time: 8.011 secs

13 tasks in build, out of which 2 (15%) were executed
11  (85%) up-to-date
 2  (15%) not cacheable
```

JLLeitschuh, 2017-01-07:
Try running a gradle clean first.

rlee287, 2017-01-07:
@JLLeitschuh Thanks, that solved my problem.

closed 2017-01-07 by rlee287