WPIRoboticsProjects/GRIP #783: Webcam source memory leak

johnyburd, 2017-01-11:
The title pretty much says it all.

[image]
[image]

Whenever I have webcam selected as the source, the memory usage constantly goes up until my computer crashes.
I'm running arch linux and this happens with or without a preview selected.

johnyburd, 2017-01-11:
I saw on the raspberry pi page that there was a bug with USB webcams, but I have also reproduced this on an HTTP stream and even a static image

SamCarlberg, 2017-01-11:
How much memory do you have installed? GRIP is configured to use up to 200MB on the Java heap but will tend to use more overall due to OpenCV's native memory use

johnyburd, 2017-01-11:
10GB

SamCarlberg, 2017-01-11:
And what version of GRIP?

johnyburd, 2017-01-11:
I also noticed the find line, contour, and blob modules all agrevate the problem.

johnyburd, 2017-01-11:
The version is the latest from GitHub

SamCarlberg, 2017-01-11:
If you go to the `Help > About` menu, what version does it report?

johnyburd, 2017-01-11:
1.5.1

SamCarlberg, 2017-01-11:
Then it's most likely a problem with Arch. I can't reproduce this on Elementary OS

johnyburd, 2017-01-11:
That is very strange. Any work around that isn't changing Os's?

SamCarlberg, 2017-01-11:
If you have a JDK installed, you can launch VisualVM to see how much of the heap is being used and how large the heap is. If the heap is small (< 50%) compared to the total memory use you see in the system monitor, then OpenCV objects aren't getting cleaned up.

thomassross, 2017-01-12:
Having the same issue on Debian Stretch

SamCarlberg, 2017-01-13:
@thomassross are you building with Gradle or using the native app from our releases page?

thomassross, 2017-01-13:
@SamCarlberg I'm using the latest deb release on the releases page.

thomassross, 2017-01-23:
I am not getting this issue with a static image instead of a webcam.

SamCarlberg, 2017-01-24:
That's expected. Changing images allocates more memory. If you're comfortable with it, you can install VisualVM and the GC plugin for it. Garbage collection should be running about once per second

NickLD, 2017-02-13:
Just a note: 

Can confirm this issue occurs with Lubuntu and fills up the 8 GB of RAM on our Vision system within Minutes.