WPIRoboticsProjects/GRIP #784: Crash opening sample grip file in 2017 vision example zip file

nollchr, 2017-01-12:
Installed latest grip and tried to open the example grip pipeline in the 2017 vision example zip file.

SamCarlberg, 2017-01-12:
How does it crash? Straight to desktop, or does a message pop up?

nollchr, 2017-01-12:
It gives a stack dump.  Let me add that

nollchr, 2017-01-12:
Thread[Project Open Thread,5,main] threw an exception
## Stack Trace:
```java
com.thoughtworks.xstream.mapper.CannotResolveClassException: grip:Pipeline
	at com.thoughtworks.xstream.mapper.DefaultMapper.realClass(DefaultMapper.java:81)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.DynamicProxyMapper.realClass(DynamicProxyMapper.java:55)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.PackageAliasingMapper.realClass(PackageAliasingMapper.java:88)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.ClassAliasingMapper.realClass(ClassAliasingMapper.java:79)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.ArrayMapper.realClass(ArrayMapper.java:74)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.SecurityMapper.realClass(SecurityMapper.java:71)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.CachingMapper.realClass(CachingMapper.java:47)
	at com.thoughtworks.xstream.core.util.HierarchicalStreams.readClassType(HierarchicalStreams.java:29)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:133)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1230)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1214)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1085)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:118)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:97)
	at edu.wpi.grip.ui.MainWindowController.lambda$openProject$5(MainWindowController.java:204)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.1
java.version | 1.8.0_101
javafx.version | 8.0.101
os.name | Windows 10
os.version | 10.0
os.arch | amd64

SamCarlberg, 2017-01-12:
And does this happen every time you open the example file?

nollchr, 2017-01-12:
Yes

vanjac, 2017-01-18:
Different versions of GRIP 1.5.1 seem to be generating different, incompatible XML files.

Some versions create a file that starts with `<grip:Pipeline>`. Other versions create a file that starts with `<edu.wpi.grip.core.Pipeline>` and only work with files like that.

SamCarlberg, 2017-01-18:
It's a bug with the library we're using to generate and read the XML save files. It defaults to use class names like `edu.wpi.grip.core.Pipeline`, but we have it set up to use more general identifiers (`grip:Pipeline`) so moving or renaming classes internally won't break save files.

@nollchr has had this problem before. Since he's the only person who's experienced this, there's something weird going on in his computer that causes the XML library to do this.

nollchr, 2017-01-18:
Lucky me!   Our team has several identical laptops with roughly same configuration.  I'll try this on a few others tomorrow.

vanjac, 2017-01-18:
What causes some installations to behave differently from others?

vanjac, 2017-01-18:
We have 2 people with the problem

SamCarlberg, 2017-01-18:
Try waiting 5-10 seconds after launching GRIP and see if this still happens.

vanjac, 2017-01-18:
Doesn't work

SamCarlberg, 2017-01-18:
Can you upload the save file to a [gist](http://gist.github.com) and share it here?

vanjac, 2017-01-18:
https://gist.github.com/vanjac/4decd254068a1942e34b4901a7cfbfcc

JLLeitschuh, 2017-01-19:
That explains the problem... The absolute paths are being used instead of the alias.

SamCarlberg, 2017-01-19:
@vanjac can you zip and upload the entire `C:\Users\<User>\AppData\GRIP` folder from a computer that's experiencing this? Something's most likely weird with the install of GRIP and I'd like to take a look.

vanjac, 2017-01-19:
I will ask one of the people...

Another problem: after the exception occurs, GRIP completely freezes. Even if this gets fixed it seems like there's a problem with exception handling.

shipshupe, 2017-01-19:
I'm one of the people, can upload later this evening

SamCarlberg, 2017-01-19:
Great.

RE: the freezes. Can you open a separate issue detailing what exactly you did to make it crash?

vanjac, 2017-01-19:
There's a 10 MB file limit, and the zipped file is 100 MB. Are there specific files that we could upload instead of the whole thing?

SamCarlberg, 2017-01-19:
No, I need the whole install. You could use google drive and share a link.

Tadtiger13, 2017-01-19:
Here you go: https://drive.google.com/file/d/0B2i21SMA63Y3LXN5UUQ0RlJkbHc/view?usp=sharing. I had put it in Program Files instead of AppData.

SamCarlberg, 2017-01-19:
I think that archive is corrupted. 7zip says it's a zip archive, not a 7z, and opening it as either is unsuccessful.

Tadtiger13, 2017-01-19:
I think Google Drive freaked when I uploaded it with drag-and-drop. This link should work: https://drive.google.com/open?id=0B2i21SMA63Y3Y19WQVNMWDdoZVU

shipshupe, 2017-01-19:
Here's another with split .7z https://drive.google.com/drive/folders/0B89gPRTYcf2ycm1PV2lGVGx4b0E?usp=sharing

SamCarlberg, 2017-01-19:
Definitely not a problem with the install. I didn't encounter your problem using your install.

Are there any user account restrictions?

@nollchr do the other laptops have this problem?

shipshupe, 2017-01-19:
no restrictions to my knowledge, can try uninstall/reinstall

shipshupe, 2017-01-19:
ah, the test file that was uploaded does work on my computer but this one initiates the crash:
[crash_test.zip](https://github.com/WPIRoboticsProjects/GRIP/files/715814/crash_test.zip)
Thread[Project Open Thread,5,main] threw an exception
## Stack Trace:
```java
com.thoughtworks.xstream.mapper.CannotResolveClassException: grip:Pipeline
	at com.thoughtworks.xstream.mapper.DefaultMapper.realClass(DefaultMapper.java:81)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.DynamicProxyMapper.realClass(DynamicProxyMapper.java:55)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.PackageAliasingMapper.realClass(PackageAliasingMapper.java:88)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.ClassAliasingMapper.realClass(ClassAliasingMapper.java:79)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.ArrayMapper.realClass(ArrayMapper.java:74)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.SecurityMapper.realClass(SecurityMapper.java:71)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.CachingMapper.realClass(CachingMapper.java:47)
	at com.thoughtworks.xstream.core.util.HierarchicalStreams.readClassType(HierarchicalStreams.java:29)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:133)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1230)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1214)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1085)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:118)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:97)
	at edu.wpi.grip.ui.MainWindowController.lambda$openProject$5(MainWindowController.java:204)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.1
java.version | 1.8.0_101
javafx.version | 8.0.101
os.name | Windows 10
os.version | 10.0
os.arch | amd64

SamCarlberg, 2017-01-23:
`crash_test.grip` still works with the install that @Tadtiger13 shared.

PR #911 (2019-03-27): Remove runtime classpath scanning
Operations, custom XStream types,and publishable classes are stored in registry files in the META-INF directory of the core project JAR. These files are generated by an annotation processor - when creating a new operation, publishable class, or save-file content, simply add the appropriate annotation to the class (`@Description` , `@PublishableObject` , and `@XStreamAlias`, respectively)

Fixes #784 
Fixes  #834
closed 2019-03-27 by SamCarlberg