WPIRoboticsProjects/GRIP #785: GRIP triggers "packager.dll not found" in Win7 64-bit.

daehlerr, 2017-01-13:
I tried to install GRIP-v1.5.1-x64.exe both as an administrator and an ordinary user.  Either way, attempting to run GRIP triggers this error message.

SamCarlberg, 2017-01-13:
Did you already have a version of GRIP installed before installing 1.5.1?

daehlerr, 2017-01-13:
No, I did not.  After the first installation failed I used CCleaner to remove it.  Tried again and failed again.

daehlerr, 2017-01-13:
McAfee reports "Prevent DLL files from Running in User Profile".  If the GRIP installer is trying to run a DLL file from my user profile, my institution's policy will block it.  I already know that cannot be turned off.  Do you have another way of installing it in Win7 64-bit that doesn't use a DLL file inside the user's profile?

JLLeitschuh, 2017-01-17:
If this was solved it would be great if you could include a description of how you fixed it.

daehlerr, 2017-01-17:
I have not fixed it.  Is that what "closed" means?  Sorry if I mis-represented the situation.  As far as I know, a different method must be used to install GRIP, which does not call a DLL file from the user's profile.  It may be necessary to build GRIP from source on my machine.

SamCarlberg, 2017-01-17:
We can configure GRIP to do a system-wide install (ie to `C:\Program Files\`) instead, but that usually requires admin privileges. I don't think it'd be possible to have the installer allow the user to choose between a system-wide or user-only install. We _might_ be able to have two separate installers... but that would get messy, fast.

daehlerr, 2017-01-17:
I do have admin privileges, but I quite agree that you don't want to have to maintain two separate installers.  I'll try out the build-it-yourself instructions and report back.  Thank.

daehlerr, 2017-01-17:
Update: building it myself as an admin seemed to go smoothly (after I installed JDK).  gradlew.bat :ui:run ran smoothly (with a lot of MissingOverride warnings) and opened a GRIP GUI window that did nothing.  I closed the GRIP window and the build console said "BUILD SUCCESSFUL".  However, when I try to run GRIP (even as an administrator), I still get the same "packager.dll not found" error message, and McAfee again tells me that it's preventing DLL files from Running in User Profile.  QUESTION: Why does GRIP itself try to run DLL files in my user profile even when I built it as an admin and am running it as adminstrator?  APPARENT ANSWER: Because the GRIP directory is created in my AppData director.  I copied it over to C:\Program Files and at least GRIP seems to run from there.  I'll now try to use it.

SamCarlberg, 2017-01-17:
Building with `gradlew :ui:run` only creates a jar file, it doesn't make a native app.

daehlerr, 2017-01-17:
I've attached a screenshot of my build session, showing that I executed gradlew.bat :ui:run (unsuccessfully because the JDK wasn't installed).  It then shows a second execution of the same command after I installed JDK, ending with "BUILD SUCCESSFUL".  After doing this, I now have a directory named GRIP that contains a file named GRIP.exe and several *.dll files -- including one named packager.dll.
[GRIP_build_log_2017_01_17.txt](https://github.com/WPIRoboticsProjects/GRIP/files/712162/GRIP_build_log_2017_01_17.txt)