WPIRoboticsProjects/GRIP #786: Generated code giving errors.

NonayMete, 2017-01-16:
When I bring up the generated grip code in eclipse I get a few errors, I am using GRIP 1.5.0 and c++

void GripPipeline::setsource0(cv::Mat &source0){
	source0.copyTo(this->source0);
}

this line shows up saying, Member Declaration not found. Eclipse also cannot find "VisionPipeline.h" and <opencv2/contrib/contrib.hpp>, despite the fact I am using wpilib 2017 and putting the .cpp and .h files in my src folder. My only solution so far is removing all these lines but I would like to know if there is a way I can fix it properly.

SamCarlberg, 2017-01-18:
The generated code isn't great and has a ton of unused imports/includes. For now, you'll have to delete the unused includes like  `<opencv2/contrib/contrib.hpp>` (the contrib module isn't included on the RoboRIO).

You should also update to 1.5.1, it fixes a few other problems with code generation.

nightpool, 2017-01-24:
duplicate of #797

closed 2017-01-24 by nightpool