WPIRoboticsProjects/GRIP #788: Microsoft Lifecam HD-3000 not starting up on GRIP in Kangaroo

derekhohos, 2017-01-16:
Using a Kangaroo (runs Windows 10) to run GRIP (version 1.5.1), the Microsoft Lifecam HD-3000 is plugged into the Kangaroo's USB port. I tried to add a webcam source to port 0. The application seems to detect the webcam as the webcam's blue light indicator blinks twice (which indicates an error), and no feed appears on the pipeline. I get a CameraSource error with the description "Camera service crashed". Then, GRIP crashes about 10 seconds later. Note that the webcam seems to work properly with GRIP on other devices, such as my laptop. I reinstalled the webcam driver multiple times on the Kangaroo, tried rolling back to previous versions of GRIP, but nothing seemed to allow the webcam feed to appear. Also note that on the Kangaroo, the webcam DOES function when I open the Camera application that comes with Windows 10 (the blue light indicator stays solid). Not sure what's not allowing the feed to work properly on GRIP on Kangaroo.

Stack trace for CameraSource error:
edu.wpi.grip.core.sources.GrabberService$GrabberServiceException: [Webcam 0] Failed to grab image
	at edu.wpi.grip.core.sources.GrabberService.runOneGrab(GrabberService.java:75)
	at edu.wpi.grip.core.sources.GrabberService.run(GrabberService.java:64)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:66)
	at com.google.common.util.concurrent.Callables$4.run(Callables.java:122)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: videoInput.getPixels() Error: Could not get pixels.
	at org.bytedeco.javacv.VideoInputFrameGrabber.grab(VideoInputFrameGrabber.java:171)
	at edu.wpi.grip.core.sources.GrabberService.runOneGrab(GrabberService.java:73)
	... 4 more

JLLeitschuh, 2017-01-17:
Can you view the webcam on your Kangaroo in another program other than GRIP?
Some other things you could try would be using different numbers.

I'm a mac user so I've never seen this before.
@saudet
This seems like its appearing in your library. Any thoughts?
For note this is what we are using for version:
```
compile group: 'org.bytedeco', name: 'javacv', version: '1.1'
compile group: 'org.bytedeco.javacpp-presets', name: 'opencv', version: '3.0.0-1.1'
```
[Link to build declaration](https://github.com/WPIRoboticsProjects/GRIP/blob/master/build.gradle#L228-L232)

derekhohos, 2017-01-17:
Yes, I can view my webcam on my Kangaroo on the native Camera application that comes with Windows 10. It appears that I have solved the issue by rolling back the Microsoft Lifecam HD-3000 driver to a previous driver version that I found off the internet. GRIP may not have been compatible with the latest webcam driver that was installed when initially plugging the webcam to the Kangaroo's USB port.

JLLeitschuh, 2017-01-17:
Strange.
Sounds like a bug in [opencv](https://github.com/opencv/opencv).

saudet, 2017-01-18:
Try with with OpenCVFrameGrabber and FFmpegFrameGrabber as well. One of them ought to work.

schwartzben, 2017-01-29:
@derekhohos What version of the Microsoft Lifecam HD-3000 driver did you install? Also, do you have the link to the website from which you installed it from? I am getting the exact same error as you and I cannot seem to find any versions of Lifecam's drivers below 3.6 (which is the most recent version).

saudet, 2017-01-30:
If the width and height are not set, 640x480 is used by default:
https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/VideoInputFrameGrabber.java#L108
So maybe that driver doesn't support this resolution?

bcteamvision, 2017-01-30:
What version of the driver did you install? We are having the same issues. @derekhohos

wjwalcher, 2017-01-30:
Having the same issue here, would love to know what version of the driver you installed @derekhohos

onenerdyguy, 2017-02-07:
Same issue here, with the default driver that installed when you plug it in.

wjwalcher, 2017-02-07:
Hey guys! Nobody has actually posted how to fix this yet, so I figured I'd resolve this once and for all. 
I was able to fix our Kangaroo by downloading the 4.25.532.0 driver from here: [http://drivers.softpedia.com/get/SCANNER-Digital-CAMERA-WEBCAM/Microsoft/Microsoft-LifeCam-HD-3000-Driver-4255320-for-Windows-10-64-bit.shtml](url). 
Let me know if you have any other questions, I'd be happy to help!

derekhohos, 2017-02-14:
Sorry for not replying! I have also figured out a solution for this:
1. Open up device manager, then expand Imaging devices.
2. Right click on the Microsoft Lifecam HD-3000, then click properties.
3. Click on the Driver tab, click Update driver.
4. Click Browse my computer for driver software. Then click "Let me pick from a list of device drivers on my computer."
5. It seems as though the 2016 driver doesn't seem to be compatible with the Kangaroo, so I installed the 2015 driver instead. If that doesn't seem to work, anything before 2016 seems to be compatible with the Kangaroo and GRIP.

derekhohos, 2017-02-14:
@saudet You mentioned earlier about setting the width and height. Is it possible to do this on GRIP so that it isn't set to 640x480 by default? The 16:9 original aspect ratio of the Microsoft Lifecam HD-3000 is converted to a 4:3 aspect ratio on GRIP, which changes the HFOV from 60 degrees (original aspect ratio) to an unknown HFOV in the GRIP image.

saudet, 2017-02-14:
Good question, @JLLeitschuh?

SamCarlberg, 2017-02-14:
We don't set the resolution at all. OpenCV seems to default to 640x480 unless told otherwise. We'd have to somehow be able to detect the maximum supported resolution and use that.

JLLeitschuh, 2017-03-01:
I think we tried setting the resolution at some point and noticed it had no effect.
Might give it a shot again though for sanity sake.