WPIRoboticsProjects/GRIP #792: Loading example grip for 2017 season

darrinjillson, 2017-01-19:
Thread[Project Open Thread,5,main] threw an exception
## Stack Trace:
```java
com.thoughtworks.xstream.converters.ConversionException: 
---- Debugging information ----
cause-exception     : java.lang.IllegalArgumentException
cause-message       : Cannot connect sockets
class               : edu.wpi.grip.core.Connection
required-type       : edu.wpi.grip.core.Connection
converter-type      : edu.wpi.grip.core.serialization.ConnectionConverter
line number         : 104
class[1]            : java.util.HashSet
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : edu.wpi.grip.core.Pipeline
converter-type[2]   : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
version             : v1.5.1
-------------------------------
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:79)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:73)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:503)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:429)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:281)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1230)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1214)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1085)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:118)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:97)
	at edu.wpi.grip.ui.MainWindowController.lambda$openProject$5(MainWindowController.java:204)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.IllegalArgumentException: Cannot connect sockets
	at com.google.common.base.Preconditions.checkArgument(Preconditions.java:122)
	at edu.wpi.grip.core.Connection.<init>(Connection.java:43)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at com.google.inject.internal.DefaultConstructionProxyFactory$ReflectiveProxy.newInstance(DefaultConstructionProxyFactory.java:126)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:111)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:90)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:268)
	at com.google.inject.internal.InjectorImpl$2$1.call(InjectorImpl.java:1019)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1085)
	at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1015)
	at com.google.inject.assistedinject.FactoryProvider2.invoke(FactoryProvider2.java:776)
	at com.sun.proxy.$Proxy8.create(Unknown Source)
	at edu.wpi.grip.core.serialization.ConnectionConverter.unmarshal(ConnectionConverter.java:60)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	... 24 more

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.1
java.version | 1.8.0_92
javafx.version | 8.0.92
os.name | Mac OS X
os.version | 10.12.2
os.arch | x86_64

JLLeitschuh, 2017-01-19:
Can you send us the file you were trying to open?

darrinjillson, 2017-01-19:
Yes. Here is the file. 
[examplegrip.zip](https://github.com/WPIRoboticsProjects/GRIP/files/717204/examplegrip.zip)

SamCarlberg, 2017-01-19:
The contents of that save file were modified in a text editor and made it an invalid save.

closed 2017-01-24 by SamCarlberg