WPIRoboticsProjects/GRIP #794: GRIP 1.5.1 Exception_Access_Violation

Teach4AZ, 2017-01-20:
Good afternoon. We tried implementing Grip 1.5.1 on a windows 10 X64 laptop. The project went fine until we tried to add the blob filter. The Grip program just closes. I can't pinpoint if it is just time or an action on the filter that causes it.  When going to %appdata% then going back a folder to local\GRIP\app there is a hs_err_pid file that says the jre encountered a fatal error. The error is Exception_Access_Violation. Anyone have ideas on why this happened and how to fix it? I read on Oracles site that it could be UAC so we turned that off and it still happens.

Axis M1013 Ip Camera
HSV Thresh
CV Erode
Mask
Find Blobs

JDK and JRE 1.8.0_121
WPILIB Software Update for Drive Station is 2016 though  Havent updated to 2017 yet.

Sam Alexander
FRC Team 1290

SamCarlberg, 2017-01-20:
Can you share the hs_err file?

Teach4AZ, 2017-01-21:
I can tomorrow.

Sam

Sent by MailWise<http://www.mail-wise.com/installation/2> – See your emails as clean, short chats.

CONFIDENTIALITY NOTICE: This electronic transmission and any attachment are the confidential property of Chandler Unified School District #80. The information in this e-mail links or attachments thereto is intended for the attention and use of the addressee only. If the reader of this message is not the intended recipient, you are hereby notified that any dissemination, distribution or copy of this communication is strictly prohibited. If you have received this communication in error, please notify the sender in reply and delete or discard this message immediately.

Teach4AZ, 2017-01-21:
Here are the files:

[files.zip](https://github.com/WPIRoboticsProjects/GRIP/files/721237/files.zip)

SamCarlberg, 2017-01-21:
Looks like GRIP can't decode the image being sent by the IP camera. Does the camera seem to work if you open the stream in a web browser?

Teach4AZ, 2017-01-21:
Yes.  The output image of the mask filter shows in grip.  Linked that filter output to blobs.

Sent by MailWise<http://www.mail-wise.com/installation/2> – See your emails as clean, short chats.

Teach4AZ, 2017-01-24:
We were able to even see the live blobs output with another project and it was working for a while until Grip just closes.  The IP camera is hooked up to the D-link DAP1522 as well as the Roborio.  The laptop is hooking up through the wifi access point.  We are using axis-camera.local as the ip address of the source.  If we are watching the live feed through Grip and then it just closes is there a way we can check to see if there was a hiccup with the feed?  Is there anything else to check if Grip closes as it is actually working?

SamCarlberg, 2017-01-24:
The crash is always happening on the thread responsible for getting images from the IP camera, and the error seems to be a problem with reading or writing to a chunk of native memory.

Can you run GRIP as an administrator and see if this still happens? If that doesn't work, try _temporarily_ disabling UAC.

Teach4AZ, 2017-01-24:
I ran at as an administrator and it still happened.  Also, the UAC is turned off through the registry.  I tried that first when I read an Oracle thread about the Exception_Access_Violation.  Are you saying native memory as in ram?  I could run memchecker to test it.